opulo-inc/lumenpnp #464: parts.csv, INDES BOM - MECH.csv & index_bom.md inconsitent, Which to use

Typhoor, 2022-02-26:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): v2.0.1
- Is this recording an active issue, or suggesting an improvement?: active issue
- Are you just logging an issue, or do you have a solution in mind?: no solution -> question

the three files are inconsistant; which is the master one, which to take ?
i.e. INDEX BOM - MECH.csv & parts.csv both show part `FDM-0026 board-holder`:
but I do not find CAD or STL files in release or on git ?!?

sphawes, 2022-03-01:
Hey @Typhoor! Thanks for the note!

The `parts.csv` file is a master list of every part number in the project. You can read about it on the Contributing page of the wiki: https://github.com/index-machines/index/wiki/Contributing#part-numbers.

'INDEX BOM - MECH.csv' is a file meant to help folks source parts if they're building their own machine from scratch.

`index_bom.md` is automatically generated from the CAD assembly, is unique to each tagged release, and is used for folks building their kit. You can read a bit more about it on the docs page: https://docs.opulo.io/docs/

I agree that these three can be confusing, and it's not clear what each one of them does at first glance. I'll rename them so that it's a bit more intuitive! Thanks for the feedback!

Typhoor, 2022-03-01:
Thanks for the explanation, actually I understood this.
Question was what is the Master, which to follow if there is inconsistency?
and ...
Where do I find CAD or 3d-files for FDM-0026 ?
Was this part abandoned ? Or replaced with a new part ? What to use as a `board-holder`

stuartpittaway, 2022-03-02:
FDM-0026 no longer exists. 

You mention board-holder, do you mean to mount the mobo PCB to the plate or to hold the work piece PCB for assembly?

FDM-9000_motherboard-mount or FDM-9007_pcb_holder_rightangle perform this function respectively.  You would need to print multiple to suit your application.

The official kits use metal standoffs instead of FDM-9000.

There is also on going work to improve the BOM, to obtain a single file, experimental version can be found here https://github.com/index-machines/index/blob/cad-staging/pnp/cad/index_bom.md

stuartpittaway, 2022-03-09:
@Typhoor can this now be closed, the latest index_bom.md file has become the single source for BOM information.

closed 2022-03-09 by Typhoor