opulo-inc/lumenpnp #466: X Gantry Assembly documentation

JesseLeonard, 2022-02-27:
Order of assembly has you put the back on first but you need access to the backside of the front part to do the belt installation. Suggest moving the installation of the backing plate and rollers to the end after finishing with the stepper motor.

JesseLeonard, 2022-03-07:
Looking at Build 2.0.1 Docs:
Steps 7, 12, 13 have you work through attaching FDM-11 and FDM-12 together with v-rollers in between.
These three steps and pictures should be moved to after step 25, i.e. after you've completed belt tensioning and any belt trimming. FDM-12 can't be installed during belt tensioning since it will be in the way of getting to the belt-side of FDM-40 and FDM-41.

The photos for steps 24, 24, 25, 25 already do not have FDM-12 installed since it has to be off in order to do z-axis belt tensioning.

Memnarch1113, 2022-05-12:
I am realizing that I did not actually address this issue. I believe it is still wrong in the docs, and we need to make it so that assembling the Z-axis stuff happens before [attaching the front and back X Gantry](https://docs.opulo.io/docs/mechanical-assembly/x-gantry-assembly/#attaching-x-gantry-front-and-back)

sphawes, 2023-05-15:
This has been fixed, all Z gantry assembly happens before front and back X Gantry assembly.

PR #533 (2022-05-02): Gabe Photo and Tooltip Pass 1

Major changes:

1. Added Tooltips with printed part serial names on them
2. Added lots and lots of CAD photos to help illustrate assembly steps
3. Fixed styling to comply with linting
4. Added Saved Camera views for the new photos
5. Clarified steps based on my personal build experience

Closes #488 
Closes #466 
(they're closed already because I manually closed them but if I had done this right this would actually close them 😓 )
closed 2023-05-15 by sphawes