opulo-inc/lumenpnp #467: Default homing steps re-order

InfiniteNesLives, 2022-02-27:
I suggest the X/Y axis' be homed first, then Z.  Changing the default HOME_COMMAND in the openpnp config to the following:
```
G28 XY; Home X and Y axes
G28 Z; Home Z axis
G1 Z25; raise Z nozzle
```

To me, it's a LOT safer to assume that the nozzle is currently at a safe Z height, than assuming there is adequate clearance at the current X/Y location to perform Z homing first.

The provided config homes the Z axis first which is quite likely to drive the nozzle into the staging plate, components, boards, etc.

I'd say so far, well over 95% of the time my machine is at a safe Z height, but not a safe X/Y location to perform a Z homing.  So it makes more sense to home X/Y first, and then Z.

it's also just more intuitive that if the nozzle is plunged down somewhere random it's prob not safe to home the machine.  By default the machine is usually left/powered down with a safe Z height, but not a safe Z homing location.  It's not intuitive that if you shut the machine down in a normal state, that you can't safely perform a homing procedure..  

I was pretty anxious about homing the machine once I was confident enough to attach the nozzle.  Had this order been flipped, I doubt I would have driven my nozzle into the staging plate/datum board so many times...

InfiniteNesLives, 2022-02-28:
Other thing I've noticed is that the first homing after power up/reset always fails for me because the ring lights don't come on until the second before openPNP starts to analyze the downward camera looking for the homing fiducial.  So the cameras don't have enough time to react to the new light and don't produce good images for visual recognition.

Perhaps there's a better solution, but adding this step to the HOME_COMMAND resolved the issue for me, so the cameras can adjust early.

```
M150 P255 R255 U255 B255   ;turn on ring lights (need camera to start adjusting before looks for fiducial)
G28 XY; Home X and Y axes
G28 Z; Home Z axis
G1 Z25; raise Z nozzle
```

InfiniteNesLives, 2022-02-28:
Sorry for the multiple separate notes! :)

But realizing one other related item I'm curious about.  For the first homing after powerup/reset the machine raises the left nozzle prior to homing (but only on first homing after power up, not on subsequent ones).

The problem I'm foreseeing, and curious about a fix, is that this also happens to drive the right nozzle into whatever rests below.  So the first homing after powerup could also cause damage with whatever happens to be attached to the second nozzle.  IMO, the behavor of first homing after power up and all subsequent ones should be identical.  The PNP can be a risky process so the mental checklist one makes before clicking that button should always be the same regardless of how long the machine has been running.

I don't have a recommended fix for this as I'm not sure what's directing this action.  My guess is that it's baked into the firmware?  I also realize this can be seen as a feature for single head users, but it's a problem for dual head users.  So perhaps a means of toggling this feature would be best?  If my assumptions are correct that it's baked into the firmware, perhaps it should be migrated to openpnp to make it easy for the user to toggle on/off as desired.

sphawes, 2022-03-01:
Hey @InfiniteNesLives! Thank you so much for all the notes! A bit of background on why we did it the way we did:

We have our Marlin config choose to home X and Y first before Z for many of the reasons you stated in your first comment. In general, moving the Z axis when the machine position is unknown is pretty prone to collision, so we prefer to home it last where we know it won't collide with anything.

We have a setting in Marlin configured where the machine moves Z to a "safe height" before homing. On boot, it thinks its position is 0, 0, 0, so it defaults to moving Z up 32mm (or somewhere around there, i forget the exact number) before homing for safety. After it homes and knows its true position, it won't move Z up because it knows it's at the safe Z height already.

When two nozzles are attached to the head, this does become a problem. Moving the left nozzle up brings the right one down, and we're back to a collision. When we support dual head LumenPnP builds, we'll actually release a separate Marlin config that takes this into account, and will likely just keep the Z axis exactly where it is where it's still pretty unlikely we'll get a collision.

As for ensuring the lights are on before fiducial homing, I believe there's an OpenPnP setting that triggers an actuator when the machine homes, I'll make an edit that includes that (or use your GCode snippet if I'm mistaken). Thanks for all the wonderful notes! I'll close this unless you feel something is unresolved.

InfiniteNesLives, 2022-03-02:
Thanks Stephen!

Yeah so I'm now realizing that most of these problems I was having are because I was using the firmware release from current github actions: https://github.com/index-machines/index/actions/workflows/build-marlin-for-index-mobo.yml

When I update to the current tagged release: https://github.com/index-machines/index/releases/tag/v2.0.1

#1: My problem with the nozzle driving into the staging plate is resolved.  (The released firmware corrects this with a similar solution I used in openpnp as you said)

#2: My concern about the right nozzle driving down on the first homing after power up is minimized if not gone because it appears the Z axis only moves ~5mm.  So the right nozzle isn't actually driven down very far.

#3: The issue with ring lights not turning on until just before imaging the homing fiducial is still present of course.  I'm curious to hear where you find the openpnp setting if you find it just for my own learning.  But yeah, quite a minor thing in any event, I just wanted to point it out as a potential issue for other users who might be confused/annoyed as to why the first homing always fails fiducial check.

closed 2022-03-02 by InfiniteNesLives

stuartpittaway, 2022-03-02:
> 3: The issue with ring lights not turning on until just before imaging the homing fiducial is still present of course. I'm curious to hear where you find the openpnp setting if you find it just for my own learning.

I've had a lot of issues with openpnp and the "async gcode driver", it doesn't appear to wait for multiple gcode commands to run/finish before moving onto the next step/movement.

Therefore, it can switch on the light and move at the same time, but often the light hasn't yet responded before the next action takes place (like the camera).

This can also be a problem with not enabling the "machine coordination" settings on the actuators.