opulo-inc/lumenpnp #471: FDM-0019 Up Light Mount Changes

AlanaCat, 2022-02-28:
- Build Number where this issue is present. (This is the same as the most recent tag number): FDM-0019-01
- Is this recording an active issue, or suggesting an improvement?: Both
- Are you just logging an issue, or do you have a solution in mind?: I have a solution
- Please describe the issue, including relevant part numbers and screenshots of the issue in CAD if possible: Depth for the ring light PCB is only 1.25mm, should be 1.6mm. There is also a capacitor that has been previously mentioned that interferes with the print, and I made som cutouts for the mouse bites so it just fits without filing.
- Please describe your suggestion solution, if you have something in mind: I have made the changes and included the .step file (in a .zip).
[FDM-0019_up-light-mount_Body_001-AC v2.step.zip](https://github.com/index-machines/index/files/8155120/FDM-0019_up-light-mount_Body_001-AC.v2.step.zip)

stuartpittaway, 2022-03-01:
@AlanaCat can I ask you to confirm that the ring light PCB was from a Lumen kit, and was 1.6mm thick ?

AlanaCat, 2022-03-01:
Yes, it is from the kit.
[image]

stuartpittaway, 2022-03-01:
@AlanaCat based on the STEP model you provided, I've made similar changes to the FreeCad model.  

This is checked into the cad-staging branch, if you get chance would be good for you to review to see if it compares with your design/PCBs

[image]

sphawes, 2022-03-08:
Merged!

closed 2022-03-08 by sphawes