opulo-inc/lumenpnp #475: Lack of Y gantry alignment means creates fluid homing fiducial

InfiniteNesLives, 2022-03-01:
Sorry for the long winded explaination/discussion here, it's mostly a dump of my discussion from discord.  It can be a bit difficult to understand the issue exactly so I figured may as well dump everything I have to explain it..

This morning my right Y stepper skipped some steps on me on its first homing after powerup.  Which sent me down a thought rabbit hole.  I recall concerns about this awhile back, but I didn’t fully grasp the implications at the time..

How does one ensure that the Y gantry is aligned with the frame/staging plate?  I don't think TMC2209 self homing isn’t an option with the shared Y stepper driver.  I was tempted to utilize a second Y limit switch.  But there’s 1-2mm of play between when the limit switch activates and the gantry collides with the leg which is too much to ensure alignment.

The implication of not knowing whether Y gantry and staging plate are parallel (or just a repeatable offset in reality) is that the homing fiducial is effectively mobile.  As the one end of the Y gantry can easily be nudged especially when powered down, it's unlikely that the Y gantry will hold the same alignment relative to the frame from one power cycle to the next.

I see that openPNP has support for a secondary homing fiducial.  I couldn’t find much about it with a brief search.  But I’m wondering if utilized and located in a far corner of the staging plate if it would provide a coordinate correction to the machines’ current gantry and staging plate mis-alignment?   (this mis-alignment is likely/viable to differ between each homing operation).

For now, thinking the easiest work around is to perform a bit of a manual start up procedure when I force the Y gantry to collide with the front legs at startup (skipping steps on L&R Y steppers), then click the homing button.

I did a quick experiment to try and measure the implications:
For the experiment, I first brought the left side of the Y gantry to touch the left leg, and then had the right side of the Y gantry ~2mm away from touching the right leg.  I then homed the machine and navigated to a location towards the right end of the staging plate where this issue will be exasperated.  Marked the X/Y location with the camera.  Then repeated with having both sides of the Y gantry touch the frame legs before homing, and going back to the same spot.

I found that with ~2mm offset in the right side of the Y gantry, I had 0.5mm error in a fixed location on the staging plate.  Not exactly massive, but with my current machine it's not hard to have the L & R sides of the Y gantry be 5mm offset from each other as that's about where it bounces back to when the machine is powered off.  With a 5mm offset between sides, my machine fails to find the homing fiducual due to my visual pipeline masking (which I don't want to adjust to support that large of an offset).

I realize this really isn't an issue with PCBs off to the side of the staging plate due to PCB fiducials.  But for everything with a fixed location relative to the homing fiducial it has an impact.

The homing fiducial does prove to effectively become mobile with this issue.  I've no clue how well aligned the L/R sides of the Y gantry have been in my past several days of machine setup.  I'm tempted to utilize my manual homing method of always starting with the Y gantry touching the front legs before homing.  And then reperform all the offsets as my homing fiducial has effectively been fluid this whole time and now I want to erase that before basing more measurements including feeders and such on that highly questionable data now.

It might be worth considering adding this manual alignment of the Y gantry prior to performing the homing procedure.  Personally anyway, I would have been valuable for me to understand this issue prior to depending on the homing fiducial as being fixed in space as it has proven not to be for me prior to using this manual Y gantry alignment method

One potential solution would be to add support for mechanical gantry calibration command G34:  https://marlinfw.org/docs/gcode/G034-mgc.html

stuartpittaway, 2022-03-02:
> This morning my right Y stepper skipped some steps on me on its first homing after powerup.

Do you know why this happened?  Does the motor have enough current?  Are the belts too tight?

InfiniteNesLives, 2022-03-02:
> > This morning my right Y stepper skipped some steps on me on its first homing after powerup.
> 
> Do you know why this happened? Does the motor have enough current? Are the belts too tight?

Yes, the right side of the Y gantry collided with the front right leg because it randomly happened to be by a sizable amount from the left side of the Y gantry for that powerup of the machine.

In other words, because there is no provided means of aligning the Y gantry with the frame, the right side of the Y gantry collided with the right front leg before the left side of the Y gantry hit the Y limit switch.  So the right Y stepper skipped steps until the left side of the Y gantry reached the Y limit switch.

stuartpittaway, 2022-03-02:
I'm also doing a manual alignment and it's working well for me too.

InfiniteNesLives, 2022-03-02:
Yeah I guess my point is, if manual alignment of the Y gantry at power up is the designed solution, it should be explained as such in the documents prior to calibrating your machine and creating a long list of locations based on the homing fiducial.  Because if you're not performing a manual alignment of the Y gantry prior to homing, I don't know how else you keep your homing fiducial from effectively being fluid moving around from one power up to the next.

If there's some other solution to this beyond the procedure and documentation, I'm curious what it is.

stuartpittaway, 2022-03-02:
I have wondered if a small spacer on the right hand front corner to place the gantry at the same offset as the left side would help.

sphawes, 2022-03-07:
> For now, thinking the easiest work around is to perform a bit of a manual start up procedure when I force the Y gantry to collide with the front legs at startup (skipping steps on L&R Y steppers), then click the homing button.

I think this is an excellent solution. I totally know what you mean, i can see how the skew would be a problem for feeder positions (or really anything that isn't being recalibrated with fiducials). In general, before I turn my machine on, i pull the Y gantry all the way to the front so that they're both mechanically up against the front two legs, then I power on and home. This has given me very repeatable results.

I am very intrigued by G34, however. This is definitely something to look into implementing. I'll close for now and add a note about this in the docs so folks don't run into this again. Thanks again for the great notes @InfiniteNesLives!

closed 2022-03-07 by sphawes