opulo-inc/lumenpnp #478: Improvement Locking PCB Through Holes

CurlyTaleGames, 2022-03-03:
This is more of a suggestion, but it'd be much easier to assemble and solder all the through hole components if you used locking through hole pads:

[https://www.sparkfun.com/tutorials/114](https://www.sparkfun.com/tutorials/114)

[image]

The holes are offset .005 inches in opposite directions allowing you to insert the part without it falling out when you flip the PCB over to solder the parts

Thanks!!!!

stuartpittaway, 2022-03-03:
or perhaps change to use surface mounted connectors, so we could PNP it?

CurlyTaleGames, 2022-03-03:
@stuartpittaway Through hole connectors/plugs are great for mechanical strength. You might be able to get away with surface mount for things that aren't plugged/unplugged often like the stepper drivers, but anything that is going to be under strain should really be through hole.

G-Pereira, 2022-03-03:
I imagine the USBs are the only ones being frequently used so no issues there. Moving away from THT seems like a better option to me as well, but there is probably some connectors that will continue to be THT due to pricing and availability so I think this can be an interesting thing to discuss.
Just out of curiosity @CurlyTaleGames have you seen this in any off-the-shelf product?

CurlyTaleGames, 2022-03-04:
@G-Pereira So I accidently did something similar on one of my PCB designs. 

I had a row of .1" header for two different parts all lined up, but the spacing between the two parts was off slightly. So when I put some header across both footprints it put some tension on the header pins holding it in place

[image]
[image]

I thought that was a really nice way to hold a part in place so I did some Googling and found that Sparkfun article.

I haven't seen this offset locking footprint on other circuit boards before, but it's something I'm going to test out with my own PCBs next time I put in an order. I imagine it would work super well for holding parts in place and then just dip soldering the entire bottom of a PCB.

I still think THT components are a good solution for plugs/connectors. I know some people in Discord have had some issues with the surface mount vacuum sensor breaking off of the PCB. I'd be really careful about what you choose to be surface mount, a PCB with ripped off pads is a lot more time consuming to fix than hand soldering a through hole part.

atanisoft, 2022-03-07:
> have you seen this in any off-the-shelf product?

I believe some of the AdaFruit and UnexpectedMaker products use the "locking" hole alignments. I'm also looking at integrating this into my projects as long as KiCad has a default footprint for it (likely they do).

CurlyTaleGames, 2022-03-07:
@sphawes no problem!

lucian151, 2022-09-16:
> This is a great idea! I agree that we should stick to THT so that we've got good strength on these connectors. I'll add this to the mobo rev04 project so it gets taken into consideration then. Thanks @CurlyTaleGames!

We ultimately wanted to do this but were unable to at this time, as this change would have broken REV04's compatibility with the REV03 pogo-pin QC Test Fixture. We'll likely update this QC fixture in the near future, so we'll revisit this suggestion then.

CurlyTaleGames, 2022-09-16:
No worries! This feature is more of a quality of life thing for soldering through hole parts, you guys do what you gotta do :)

sphawes, 2023-05-15:
Going to close this as we'll likely stick with inline with the mobo for now, but @CurlyTaleGames thank you for the idea! It actually got integrated into the feeder for the programming port so no connector is needed to flash new firmware to the feeder!

closed 2023-05-15 by sphawes