opulo-inc/lumenpnp #483: Your Title Here

InfiniteNesLives, 2022-03-07:
NOTE: this issue is still a work in progress.  It was requested that I go ahead and submit a github issue on this problem prior to diagnosing the issue as much as I'm able.  I have more plans for testing and diagnosis that I will update as my efforts continue.

Please provide information about the board with the issue:
- Is this issue based on a BETA Board, or an actual release motherboard design?:  Release mobo self assembled.
- What is the revision number of your board, including the commit number?  "REV03 156fadc"
- PC info: Dell XPS13 laptop, windows 10 pro, i7-8550U @ 1.80Ghz w/8GB ram.
 
Talk about the issue you're having here: 
- What isn't working?  USB communications proving unstable, will get 'USB device disconnected chime' from windows in the middle of a job about every 10-30min.  Sometimes using disconnect/reconnect will recover, sometimes a power cycle/reset is required as the mobo won't respond anymore, sometimes windows requires the device to be fully uninstalled, powercycled, and reconnected like a new device to regain communications.

- What does it prevent or degrade?  Much difficulty/annoyance to have the mobo stop responding to USB comms in the middle of a job & then recover, re-home, check fiducials again, recover feeders or parts that are now mis-picked due to machine dropout mid-job.

- What have you done to diagnose that it is a hardware issue?  I'm still working on this.  So far I've tried different USB cables.  I'm also not plugging anything else into the USB-A sockets on the mobo (like cameras).  I was having issues with my one camera that I was connecting to the mobo hub dropping out of USB comms routinely, which was even more of a pain to regain in openpnp.  So I connected all my cameras directly to my PC, and they have been fine since.  So I'm highly suspect of the USB hub on my mobo as a potential source of the problem.

- Attach supporting files like software logs / scope images / logic captures / annotated schematic or pcb drawings.
openpnp log attached with multiple read errors.
[mobo usb dropouts.txt](https://github.com/index-machines/index/files/8198381/mobo.usb.dropouts.txt)

Future testing plans:
Figure out if PC problem or mobo problem by:
-try another PC, maybe try a linux based machine instead of windows
-try my the mobo from my opulo kit to see if having same problem, or if just a problem with the mobo I assembled.

If mobo problem:
-try removal of the USB hub IC from my mobo to half split between a problem with hub IC or the STM32.

stuartpittaway, 2022-03-07:
Can you confirm what power supply you are using with the PNP ?

InfiniteNesLives, 2022-03-07:
> Can you confirm what power supply you are using with the PNP ?

Ah yeah that is one other item I want to test and forgot to mention earlier.  I'm currently using one of my bench power supplies.  I'm not terribly suspect of it being the cause since the problem frequently occurs while my machine is idle with a constant and low power draw.  But yes, power delivery it is certainly a potential cause for this issue that I need to scrutinize.

bholmes451, 2022-03-07:
I would also check your Windows Power Settings. Specifically the USB Selective Suspend setting in the attached image. Our software uses a USB Dongle but Windows will try to put the USB port to sleep even when we are sending data.

[image]

InfiniteNesLives, 2022-03-07:
Yeah I agree that's a good thing to check.  It was recommended to me in discord, but I simply don't have any USB settings available in my power options though.

bholmes451, 2022-03-07:
I find it in Windows 10 by pressing the Windows Button and then typing 

edit power plan

I then select the best match which opens the "Edit Power Plan" control panel. You then have to click the "Change advanced power settings" link to open up the window.

sphawes, 2022-03-07:
@InfiniteNesLives You didn't mention this in your initial notes, but has the STM32 ever reset as part of this (ring lights go through an RGB sequence)? or does Marlin stay running, it's just communication that dies out?

InfiniteNesLives, 2022-03-07:
@bholmes451 I can find advanced power settings just fine.  Not all computers have "USB settings" though it appears.  "Balanced" is also my only option in the pulldown FYI.
[image]

@sphawes In all the times it's happened, I have not seen the STM32 reset itself with ringlights going through their RGB sequence.  Usually the COM port is still listed in device manager & devices and printers.  One time it failed extra hard though and windows reported that the mobo's device descriptor failed.  power cycling didn't fix that case.  I had to completely uninstall and remove the mobo from windows and then power cycle to recover that time.

So on the whole, when the problem does occur, it's acting as if the STM32 isn't responding to commands.  But no clue where in the windows -> USB -> cable -> mobo hub -> STM32 chain that problem is occuring.  Windows usually gives indication that something happened to a USB device at least half the time windows gives the "USB device removed" chime, but not always.

bholmes451, 2022-03-07:
Wow, I have never seen an advanced settings dialog box so empty outside of a corporate user network. All modern machines have the capability. The only way I know to remove those settings is to limit them at the Network Administrators level.

InfiniteNesLives, 2022-03-07:
Yeah IDK, I did purchase this laptop as an open box back in ~2017.  Perhaps somethings like that were tweaked by said box opener.  I will be trying other computers and things to ensure that's not the problem.  But considering I'm getting the "USB disconnected" chime, I don't heavily suspect power settings to be my issue.  But the main reason it was requested that I create this github issue is so that it could be used as a future reference for others, and the power settings is certainly a good  thing to check first.

sphawes, 2022-03-07:
Hmm alright. Looking forward to hearing how debugging goes!

In regards to this:
> One time it failed extra hard though and windows reported that the mobo's device descriptor failed. power cycling didn't fix that case. I had to completely uninstall and remove the mobo from windows and then power cycle to recover that time.

What did you do to "uninstall and remove the mobo"? Do you mean delete drivers and such? I'm mainly on mac/linux, so would love to know what you did there so we can help other folks that might have the device descriptor issue.

InfiniteNesLives, 2022-03-07:
First easy option is to open "devices & printers" in the control panel, find the device (usually towards the bottom), right click on it and select "remove device"  Can see the option in my first picture but COM3 is working here, when it's device descriptor has failed it'll look like my second image: 
[image]
[image]

I think the more thorough option is to uninstall it from the device manager though, find the device and right click to find uninstall option.
[image]

Off the top of my head, I don't recall if these exact options are available to you when the device descriptor has failed.  So these may not be the best instructions.  But these are the 2 place I go to find what options I have to uninstall/disable/remove the device.  If/when I see it again as device descriptor failed, I'll better document my process to share with others.

From there, you want to remove the USB cable from your PC, and fully shut down the lumen power.  Might not even be a bad idea to restart your PC.  Real goal is to make windows think it's a brand new device and have it go through it's normal new device setup next time you plug it in.  You also want the USB hub on the mobo to fully restart so that's why you should remove all power to the lumen too before reconnecting via USB.

InfiniteNesLives, 2022-03-20:
Problem turned out to be nothing more than a bum USB cable I guess.  Swapped my original cable I've been using for years with various projects for the one in my opulo kit and it's been working fine.  I swapped back to the old cable and it dropped out again in a matter on mins confirming the problem.

closed 2022-03-20 by InfiniteNesLives