opulo-inc/lumenpnp #484: Bottom camera focus distance would drive a 2nd nozzle into staging plate

InfiniteNesLives, 2022-03-07:
Sorry IDK how to classify this issue, I would say it's overall design issue, but it could be solved with cad or perhaps documentation, or maybe I'm just misunderstanding something.  So mostly just wanted to bring this issue up for discussion.

Opulo bottom camera position calibration docs prescribe a Z=61mm focus distance in order to give adequate field of view for larger components:
https://docs.opulo.io/openpnp/calibration/bottom-camera-position/

The problem with this height is that when we get to the point of adding anything to the second head, that second head will have to be driven through the staging plate to get the part in the first head to focus.  

Perhaps this wouldn't be an issue if the second head is used for pasting and it can have a ~10mm offset from the nozzle since a pasting tip should never have to reach below ~10mm above the staging plate.  But for those planning to use the second head as a PNP nozzle (myself included), this is a significant problem.

Brainstorming some potential fixes:
-Switch from 6mm camera lens to 2.1mm to give adequate field of view with a Z height of ~30mm where both nozzles would be at equal height.  This isn't really a viable option for those with 6mm cameras provided by opulo.  I also expect there is a reason that the design switched from 2.1mm lens cameras to 6mm, perhaps for reduced fish eye distortion.  But for those yet to acquire their cameras, this could be considered.  (I have 2.1mm cameras on my current machine as it was sourced based on the BOM prior to release).

-Modify bottom camera mount to lower the camera by ~30mm to give adequate field of view with 6mm lens camera with a Z height of ~30mm.  This would have to be tested of course, but potentially easy CAD fix & reprint.  The biggest downside of this solution (especially if not deployed soon), is that everyone will likely have to revisit each and every openpnp vision pipeline for their bottom camera as lighting and other nuances of the bottom image may change.  Perhaps not though, would have to do more testing.  But if this is the planned solution, there would be great benefit to users to deploy this fix asap before too many people have tuned all their bottom vision pipelines.

-Modify suggested Z height of focus and reduce field of view.  IDK how far this approach should be pushed as the loss of field of view could be a deal breaker.  But maybe ~5mm isn't too big of a loss to keep a second nozzle + component from colliding into the staging plate?  The biggest 'problem' with this solution is that users are very likely to have to readjust many of their bottom vision pipelines because the effective size of their components will be changing.  This would be a considerable amount of effort depending on how many different components the user has.  If this is the chosen solution, there would be great benefit in deploying it asap to minimize user impact.  Doesn't appear too many kit purchases have reached this step yet, but that's likely to change very soon as earlybirds have had their kits for a few weeks now.

stuartpittaway, 2022-03-08:
@sphawes what do you think about changing the CAD model for the up-facing camera/bracket - we could lower the camera 10mm below the base plate to remove the need to go to -61mm which is way too far out.

sphawes, 2022-05-17:
I think that lowering the camera by 30mm would solve this problem excellently. To your point @InfiniteNesLives, I think lots of setup would transfer over well to a different camera height as long as pixels/mm is updated, but things in the pipeline that use pixels as opposed to millimeters would have to be updated.

I think the tradeoff of folks being able to just reprint a part and do some software tweaking is better than needing a new lens and doing some (lesser) software tweaking.

Going to rename this so the issue title represents the fix that's needed!

stuartpittaway, 2022-05-17:
This will also need to tie into Lucian's changes for the bottom foot under the stage plate.

InfiniteNesLives, 2022-06-01:
I ended up dropping the 2mm lens camera & ring light on my machine by 13mm so that I would have good field of view with components at the same height they're being placed on the PCB thus negating any inaccuracies when the Z axis isn't perpendicular to the X/Y plane and also coincident with the bottom camera line of sight.

In order to permit using the same bottom camera mounting bolts (avoiding the need for super long 50mm+ bolts), I had the spacer twist the ring light & camera assembly by 45 degrees.  Then I had to modify the camera mount part to twist the camera 45 degrees back again.  The original camera mount could be designed to support the bottom camera in both rotations with the same print to avoid needing to reprint if performing this modification later on.

I also integrated jazzychad's screw foot modification into the design.  My exact modifications probably aren't of much use unless one has a 2-3.6mm lens in the bottom camera.  But I wanted to go ahead and share my approach to dropping the camera.

I made a little camera focusing tool too, but it needs some beefing up because it warps quickly when it warms from the camera & ring light heat.

[13mm camera drop.zip](https://github.com/index-machines/index/files/8817899/13mm.camera.drop.zip)

[image]
[image]

PR #606 (2022-09-16): V3 CAD Changes + Dual Nozzle Config
## Summary
This PR contains the CAD updates that uprev the LumenPnP to v3, and will be part of the v3.0.0 release. See the list of changes below for all the details. 

## Changes
- Removable X Axis
  - The entire X axis subassembly is now completely removable from the Y gantries using only four screws.

- Cable Chain replaces Umbilical
  - A cable chain now routes all pneumatic tubing and cabling to the head from the front left leg.
  - A new part has been added that bolts into the front left leg that supports the cable chain, and the X gantry has been modif [... cut]
closed 2022-09-16 by sphawes