opulo-inc/lumenpnp #491: Vacuum sensing test missing step to actuate valve

InfiniteNesLives, 2022-03-17:
https://docs.opulo.io/docs/testing/actuator-check/#vacuum-sensors

The vacuum sensing test in opulo docs never instructs you to actuate (power on) the vacuum solenoid valve.  Unless I'm mistaken, this is a vital & required step to draw vacuum on the sensor.  Without powering the valve, the line is vented to atmospheric pressure.  I don't have vacuum sensing enabled on my machine so I could be wrong, but several users are reporting problems with the vacuum sensing test and I think this is probably why.

atanisoft, 2022-03-17:
I'm working on configuring OpenPnP for my projects and after importing the Opulo machine.xml (via overwrite in config directory) I found there are a couple Actuators that are defined on the GCodeDriver that don't exist:
https://github.com/index-machines/index/blob/01702144b94ddbe14eabd6cf24d7e3c43b42dba0/openpnp/machine.xml#L253-L263

I haven't dug into the Marlin code to see exactly what these codes do in regard to the Opulo but I suspect this may be related to the vacuum sensing.

jellewie, 2022-03-18:
Just looked up the Gcode of @atanisoft , just here to report them:

https://marlinfw.org/docs/gcode/M042.html Set Pin State - M42 Ignore Pin State Mode
https://marlinfw.org/docs/gcode/G004.html Dwell - G4

https://marlinfw.org/docs/gcode/M106.html Set Fan Speed - M106 MaterialIndex index speed secondary

https://marlinfw.org/docs/gcode/M150.html Set RGB(W) Color  - M150 intensity pixel intensit intensity strip intensit intensity

atanisoft, 2022-03-18:
Yeah, I don't know if these actuators are important or not but they are defined on the GCodeDriver but the `head-mountable-id` value is invalid for any of the defined actuators.

stuartpittaway, 2022-03-18:
> The vacuum sensing test in opulo docs never instructs you to actuate (power on) the vacuum solenoid valve.

It may be missing from the documentation, but Stephen was using these two commands when the pump gets switched on.  This means both the pump and the valve operate together, which is how he's been running.

There were a few comments about OpenPNP not opening/closing the valve at the correct point in the pnp process if the valve was a separate actuator.  I've tried both and didn't see a problem.

"Pump Left" actuator GCODE ```ACTUATE_BOOLEAN_COMMAND```
```
{True:M106 P1}{False:M107 P1}
{True:M106 P0}{False:M107 P0}
```

stuartpittaway, 2022-03-18:
> https://marlinfw.org/docs/gcode/M042.html Set Pin State - M42 Ignore Pin State Mode

Probably don't need those pin states as the outputs on the mobo are connected to the fan header/pins, so M106 can be used.

InfiniteNesLives, 2022-04-07:
Bumping this issue as people are still actively running into this problem..

sphawes, 2022-05-11:
@InfiniteNesLives Thanks for reporting this, definitely incorrect on the docs side; should have the extra line to actuate the valve as well. This will be fixed with the next documentation release we're prepping for now!

commit 36d5fd80 (2022-05-17): fixes #491 by instructing the user to also actuate the valve when testing their vacuum sensor. also updates the command for the vacuum sensor to represent the update to Marlin's M3426 command
closed 2022-05-17 by sphawes