opulo-inc/lumenpnp #492: Populating the Staging Plate

KohalaJim, 2022-03-18:
The datum board and mount is shown attached to the staging plate in the final image of this section, however, there's no textual instruction citing the fastener choices that are recommended.

jonnerd154, 2022-03-19:
@KohalaJim You actually don't want to mount it during assembly. In the OpenPnP section of the Docs, you hold the datum board in your hand, showing the array of fids on bottom side of the board to the camera for fisheye compensation. After that, [the instructions](https://docs.opulo.io/openpnp/calibration/set-camera-exposure/) direct you to bolt the datum board to the staging plate.

I agree that image at the end of [this page](https://docs.opulo.io/docs/mechanical-assembly/populating-staging-plate/) is a bit confusing, but the textual instructions are good. I think most of us are Lego people and want to make what the picture shows!

Memnarch1113, 2022-04-26:
I added a note that says to disregard the datum plate until later

closed 2022-05-02 by Memnarch1113