opulo-inc/lumenpnp #495: OpenPNP backlash compensation and calibration error

InfiniteNesLives, 2022-03-20:
[image]

Both myself and JelleWho are getting this error for both X&Y axis.  Looked around for the settings the error is complaining about but couldn't figure out what to do.  Anyone else figure out the solution for this error?

jellewie, 2022-03-20:
Just to note, I have used the latest [machine.xml](https://github.com/index-machines/index/blob/01702144b94ddbe14eabd6cf24d7e3c43b42dba0/openpnp/machine.xml) from the repo (and all other setting). 

This exact issue doesn't ocured when I tried to set up my own profile. To me it seems to be something configured wrong in openPnP

InfiniteNesLives, 2022-04-28:
I don't have time at the moment to drop all the details here, but JonMoyes sorted out a solution to this problem and we were both able to sucessfully complete the X & Y automatic backlash calibrations: https://discord.com/channels/688723670461448225/814870897722523699/969018146566570005

Still not sure what to recommend for speed and acceleration settings which are a big question to answer before really utilizing the backlash calibration from what I would guess..

InfiniteNesLives, 2022-05-09:
After following Jon Moyes instructions I realized the suggested acceleration was quite slow.  My machine was painfully slow compared to beforehand.  Idk what acceleration settings are recommended, but I ended up picking 500/s^2 on Y, and 800/s^2 on X.

G-Pereira, 2023-08-30:
Machine motion is getting significant changes. Please reopen if this is still an issue on 3.1.0+ release.

closed 2023-08-30 by G-Pereira