opulo-inc/lumenpnp #503: Incorrect quantity in BOM

boyddotee, 2022-03-29:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number):
- Is this recording an active issue, or suggesting an improvement?:
- Are you just logging an issue, or do you have a solution in mind?:
- Please describe the issue:

OTS-00144, listed as quantity 4 at the following location should it not be 2?

https://github.com/index-machines/index/blob/master/pnp/cad/index_bom.md

- Please describe your suggestion solution, if you have something in mind:

Change to 2?

Memnarch1113, 2022-04-26:
@sphawes is this file auto-created or can I just change it? `OTS-0014` we really do only have 2, not 4

sphawes, 2022-06-06:
fixed with CI updates! thanks for the issue @boyddotee!

closed 2022-06-06 by sphawes