opulo-inc/lumenpnp #509: Mobo Part Silkscreen Improvements

qwertymodo, 2022-04-10:
It would aid DIY assembly of the mobo a lot if the part footprints were updated to use clearer pin-1 indicators, such as dots in the relevant corner, instead of the current "one line is longer than the other" marking style.  In a lot of cases, it's very difficult to differentiate, and I'm finding myself searching for photographs of a populated mobo, or tracing out pin continuity on several parts to verify the correct part orientation.  Also, there is no silkscreen at all for the alternate vacuum sensor footprint on the bottom of the PCB.

TGiles1998, 2022-05-03:
Hi, same here, did u find a good image of the assembled board? Its a bit of a nightmare

TGiles1998, 2022-05-03:
[image]

This is the image i used, i could just make out the orientation markings on the components.

atanisoft, 2022-05-03:
There is an interactive BOM html file in the releases that has all silk screen, components, etc all easily viewed on the PC.

qwertymodo, 2022-05-03:
The interactive BOM is great, but it doesn't help the silkscreen pin-1 indication issues that were the point of this post.

atanisoft, 2022-05-04:
> but it doesn't help the silkscreen pin-1 indication issues

Unfortunately, KiCad only includes footprints that have a longer line for the pin 1 indicator. Short of creating custom footprints or adding many items to the silk screen I don't know how this can be improved unfortunately.

TGiles1998, 2022-05-04:
Hi, Could i have access to the KiCad and I am happy to add the dots to the silkscreen as this will form part of my process for ensuring the board I'm building will be correct before I apply power. This is my current stage before putting it in my reflow oven.

[image]

Thanks,

Thomas

G-Pereira, 2022-05-04:
Be aware that STM32 should be rotated by 180 degrees. As the OP said, the pin 1 is currently marked with a long silkscreen line close to it and the IC has a round indent on the plastic casing near the pin 1

G-Pereira, 2022-05-04:
@TGiles1998

TGiles1998, 2022-05-04:
@G-Pereira Hi,

Could I have access to the Kicad files so I can place the dots on the silk screen. The orientation of the diodes and SOP-8 modules are also very hard to determine, along with the orientation of the crystals as well.
Thanks,

Thomas

G-Pereira, 2022-05-04:
Of course, they are right here in this repository.
Go [here](https://github.com/index-machines/index/tree/master/pnp%2Fpcb%2Fmobo)

closed 2023-05-15 by sphawes