opulo-inc/lumenpnp #514: "Build Number" dropdown order is confusing/badly ordered

jazzychad, 2022-04-13:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): all

- Is this recording an active issue, or suggesting an improvement?: active issue
- Are you just logging an issue, or do you have a solution in mind?: solution
- Please describe the issue:

Please see this screenshot of the current doc site header nav:

[image]

This order of build numbers is somewhat confusing because they are numerically out of order, and as more build numbers are added, it will become more confusing to know whether you are selecting the most current version for any particular `X` or `X.Y` or `X.Y.Z` version if they are well -ordered.

- Please describe your suggestion solution, if you have something in mind:

The build numbers should be ordered. In the current example from the screenshot, I would expect the following order:

```
2.0.0
2.0.1
2.1.0
```

sphawes, 2022-05-17:
Thanks for this @jazzychad, i agree that it's confusing as heck.

@daveismith how is Hugo organizing these? From poking around a bit, i couldn't see a flag for ordering style in Hugo. I'm guessing it's just dropping in a new one each time we release, but not sure how to ensure they're put in a correct order.

sphawes, 2022-06-06:
we've switched the docs to not have specific versions for now! the granularity we had with all these different releases turned out to not be that useful. when we need to reimplement, we'll make sure we can have them ordered properly. thanks for the issue @jazzychad!

closed 2022-06-06 by sphawes