opulo-inc/lumenpnp #515: Calibration print (FDM-8000.stl) is not in 3d-prints.zip

jazzychad, 2022-04-13:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): 2.1.0
- Is this recording an active issue, or suggesting an improvement?: active
- Are you just logging an issue, or do you have a solution in mind?: solution
- Please describe the issue:

See page for reference: https://docs.opulo.io/tag/v2.1.0/docs/mechanical-assembly/printing-parts/

Instructions say only to download `3d-prints.zip` from the assets of latest tag (in this case, 2.1.0). The next step says to "Before printing the parts, print the calibration print. This is FDM-8000, ..."

This file is not included in the `3d-print.zip` archive, and it took me a while to realize it was actually a separate file in the assets list.

- Please describe your suggestion solution, if you have something in mind:

Either include this file in the `3d-prints.zip` archive, or deliberately specify in the documentation to download the `FDM-8000.stl` file separately, or both.

atanisoft, 2022-04-16:
@jazzychad I agree it would be great to have this in the zip file rather than only on the release itself. It would require the FreeCAD file to be checked in and referenced as part of the actions.

sphawes, 2022-06-07:
yup, this is really about getting the corndog modeled in FreeCAD. @SyberxSpace this would be a great foray into FreeCAD!

lucian151, 2022-09-07:
@bricefekete is working on this! Reassigning to him.

lucian151, 2024-05-22:
We will no longer be doing this as the LumenPnP comes pre-printed by Opulo.

closed 2024-05-22 by lucian151