opulo-inc/lumenpnp #522: Version-specific banner in documentation is unnecessarily ominous

jazzychad, 2022-04-18:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): all
- Is this recording an active issue, or suggesting an improvement?: active
- Are you just logging an issue, or do you have a solution in mind?: soution
- Please describe the issue:

See a hardware-specific documentation page, for example: https://docs.opulo.io/tag/v2.1.0/docs/

At the top is a warning banner that look like this and says: 

> Version v2.1.0 of the documentation is no longer actively maintained. The site that you are currently viewing is an archived snapshot. For up-to-date documentation, see the [latest version](https://index-machines.github.io/index).

[image]

This wording makes it seem like this documentation is out-of-date and could be considered wholly _deprecated_ .. which I think is overly strong wording, because (I think) you are _supposed_ to be looking at hardware version specific docs for your machine while assembling it.

On the first page of the docs, there is a giant warning section about making sure you know your hardware build number version: 

[image]

It ends with the following text:

> Always use the most recent, so long as the first two numbers match your hardware

So, I think this warning, combined with the wording in the banner when looking at hardware-specific pages will cause some confusion (it certainly did for me). In my opinion the wording should be changed to be less "this documentation is unmaintained, please stop reading them" and more "these docs are for build number X.Y.Z, if that's your build number, then please continue!" (see below)

- Please describe your suggestion solution, if you have something in mind:

I would suggest changing the language in the version-specific banner to something more like:

> You are viewing the documentation for Build Number {X.Y.Z} - Please make sure you are viewing the documentation for your machine's build number. If this is the incorrect build number, please select your build number from the dropdown in the navigation.

Having a link in the banner to the "latest version" is also confusing because it is an untagged (main-branch?) version of the docs which are not tied to _any_ hardware build number, so who should use them??

sphawes, 2022-05-17:
@jazzychad I think there's a pretty straightforward flag in Hugo that just completely removes this banner altogether. the mega warning on the first page does a pretty good job of informing folks about the versioning system. for now I'm just going to yeet this banner altogether.

Moving forward, we're going to be changing the versioning system a bit; it's confusing as hell right now, but that's mainly in an effort to have people easily know what changes 1) are just a software update, 2) require a reprint, or 3) require different hardware than what came in their kit. This is pretty idealistic, and in reality not very easy to use. Actively thinking about how to make this better.

sphawes, 2022-06-06:
This has been fixed! version drop downs and the spooky banner are no more. we'll reintroduce versions again when needed, and when that happens we'll avoid the spooky banner. thanks for the issue @jazzychad!

closed 2022-06-06 by sphawes