opulo-inc/lumenpnp #536: Vacuum/Pressure Sensor Tore off (THT)

Memnarch1113, 2022-05-04:
[Ticket 806200619](https://app.hubspot.com/contacts/20964002/ticket/806200619/)
Motherboard REV03

Customer Service emails about the vacuum sensor tearing off the motherboard during assembly/use. I believe the plan was to replace the THT sensor with an SMD one, not 100% sure if that work was already done or not.

AlanaCat, 2022-05-13:
For the issue of pads coming up on SMD components with possible tension on them, I would like to suggest this technique, expanding the pads under the solder mask, which gives them much more "holding strength" on the PCB.
[image]
.

atanisoft, 2022-05-13:
larger pads would certainly help in this as well ensuring the parts have the proper squish/paste. In my case VAC2 was not sufficiently squished into the paste and did not reflow fully as a result. During testing of the MB at the I2C step I found errors and tilted the machine onto it's back feet and spotted that VAC2 had came off and was barely hanging on by one pad. When I removed the PCB from the machine to get a closer examination on the damage to VAC2 it fell off entirely and unfortunately took the pad and part of the trace with it.

This issue should be renamed to drop `THT` and instead be for `SMT` as the VAC2 sensor is `SMT` on REV03 MB.

In REV04 the VAC sensors have been swapped for THT last I checked which will also help resolve this issue.