opulo-inc/lumenpnp #540: Need more info on Limit Switches

solarmax1, 2022-05-07:
Please fill out the following:
- Build Number where this issue is present. (This is the same as the most recent tag number): v2.1.1
- Is this recording an active issue, or suggesting an improvement?: Improvement
- Are you just logging an issue, or do you have a solution in mind?: Have solution in mind.
- Please describe the issue: The documentation does not cover wiring & testing the limit switches.  If someone has an Opulo kit, it would be obvious, but it you are building your own Lumen, you are on your own.

- Please describe your suggestion solution, if you have something in mind:
1. I could not find a mention of hooking up the YL limit switch.  It is possible I just missed it.
2. The testing of the actuator section should describe how to read the limit switches from Marlin.  That way they can be tested before homing and the head crashes into something.
3. The wiring of the limit switch cables is not obvious.  If you have an Opulo kit it is, but if you are building your own there should be a diagram, schematic, and photo of what connects to what (3-pin mobo connector to 4-pin limit switch connector)

sphawes, 2022-05-17:
hey @solarmax1 thanks for the issue! i agree on all these points. I've just fixed 1 and 2 and they're currently in the doc branch. The third one is a bit more involved, so im going to rename this issue to reflect that and get some wiring information in the repo! thank you!

sphawes, 2022-06-06:
Hey @solarmax1! I've added wireviz yaml files to the repository: https://github.com/index-machines/index/tree/master/pnp/cad/CHA. Closing this!

closed 2022-06-06 by sphawes