opulo-inc/lumenpnp #544: Add Previous/Next buttons on the documentation website

Vultik, 2022-05-15:
Please fill out the following:
- v2.1.2
- Suggesting an improvement
- Add Previous/Next button on the documentation website
- It will be very convenient to have a next button on the bottom of each page, to acces the next page.
- Example :
[image]

sphawes, 2022-05-17:
@daveismith Seems like there might be support for this in Docsy based on [this](https://github.com/google/docsy/blob/main/layouts/partials/pager.html), but I'm not sure how to go about enabling it. Am I correct in that this is something we could just turn on?

theacodes, 2022-05-27:
That pager is more intended for blog posts (where there's a clear next/prev). Most Docsy examples just have a "Next steps" section at the bottom. I can send a PR to add this to the guides. :)

sphawes, 2022-06-06:
This has been merged in the new docs repo due to @theacodes PR fixing it. Thanks for the issue @Vultik!

closed 2022-06-06 by sphawes