opulo-inc/lumenpnp #545: Mobo REV4 improvements

ZOUDIO, 2022-05-16:
After some improvements that did not affect lay-out (https://github.com/index-machines/index/pull/472) I also have some (optional) improvements for the mobo that change connectivity and/or affect lay-out:
1) Recently it was mentioned by Stephen in Discord that the mechanical support pins on the USB-C are soldered by hand. These pins can be added to the paste layer, so they are already soldered during reflow (pin-in-paste)
2) Is there any benefit to supporting multiple stepper drivers besides sourcing? It complicates the lay-out quite a bit, takes up boardspace, and adds to the complexity of the product. Of course it increases the flexibility, but is this beneficial? Are different stepper modules suited for different usecases? Or is it just the a different route to the same goal?
3) Since RE/DE on the RS485 driver are normally toggled together, they can often be combined.
4) Why were bottom fidicials added if there are no components on the backside?
5) Maybe the DIN symbol is already considered temporary, but it gives a wrong impression (that it's a single row header)
6) Same with the new symbols in the vacuum_sensors sheet. They are fine electrically, but a bit different to the KiCad convention.
7) Why is DIN choosen? Is it to switch to a laptop style power supply?
8) U13 is missing input decoupling. 
9) Power plane routing is quite sub-optimal now. A common stackup for this purpose would be signal/ground/power/signal. Dedicating a layer to power cleans up the lay-out, both visually and noise-wise.
10) Im currently looking around a bit, but it would be nice to have vacuum sensor with an adress pin. This removes the need for the I2C multiplexer. The THT pad drill on this footprint seem terribly small, has it already been tried?
11) Grid origin seems to be set in a random point in the upper left quadrant of the board. Bottom left corner seems like the appropriate place.
12) Watch out with the USB_HUB_OVERRIDE jumpers under the USB hub IC. I get the idea, but if you order a stencil now, you will have paste under your IC if populated. Something to consider.
13) U11 is quite close to its gate voltage, I would add a zener clamp or at least a voltage divider.

And some personal opinions, do what you want with them :) I can also see why you would do it the way it is currently.
14) Because normally symbols are grouped by value, their function is better described in seperate field. For example, currently the limit switches show up as differnent parts in the symbol field editor, even though they are the same part.
15) The schematic is not really the optimal place for purchasing numbers. Ideally it only conveys the partnumber and an external tool like Octopart can scrape live purchasing data.
16) The lay-out is crammed into 100x100mm presumably for cost reasons, but in practice it will barely matter and the main costdriver is surface area. The components seem to fit much better in a rectangular lay-out.

If there are any points that you want to outsource to me, let me know. Then I will fix it and create a pull request.
Great to see this project moving along rapidly, I will probably get a kit end of summer!

sphawes, 2022-05-17:
Hey @ZOUDIO! Thank you for all the insightful feedback! I'll address your points:

1. We've tried soldering these with reflow before, and we weren't happy with the results. It's currently a post-reflow step, and doing so really lets us make sure they're on there good. We really want to make sure the problem of connectors breaking off is solved thoroughly, even if it takes an extra step.
2. We have the stepper drivers be removable so that people can hack the board and modify it with other drivers if they'd like, but we only ship one kind of driver (TMC2209, set up to work with the default jumper config). we also like the fact that if someone blows out a driver, they can get up and running with just one replacement driver instead of the whole motherboard.
3. We considered this when doing the routing to the MAX3078E chip, but given the fact that we have plenty of extra pins, we decided to break it out to two. Not sure if there are any pin-compatible replacements where these pins _cannot_ be connected, but we figured better safe than sorry!
4. Great question! At some point, we'd like to have a LumenPnP head automatically do the THT soldering, and having backside Fids lets us do this. Far down the road, but wanted to get them in there for testing and playing around!
5. This was a temporary bodge for sure, the correct move is to make a proper symbol for it. Just an artifact of me wanting to get a Beta board built up for testing ASAP. Should be fixed.
6. Same thing as #5, just a placeholder!
7. We chose this connector because 1) we can get off-the-shelf power supplies for it, and 2) it is able to carry the current needed for the machine. A typical barrel jack doesn't suffice. This is actually the same connector used in the Prusa Mini 3D printer.
8. Good catch! I think I added them based on the schematic, but they're just in an unideal location, and probably on the wrong side of the 5V multiplexer diodes.
9. Yeah, I had to use a bit of the dedicated 3v3 power plane for routing. Is it better to do this on the GND layer instead? I figured keeping the GND layer solid and consistent was more important than 3v3.
10. I totally agree, we've been looking for one from CFSensor, but weren't able to find one. If you happen to find one with a configurable address, please let me know! And I believe I followed the datasheet for the pin diameters, but could have messed it up. We get vacuum sensor samples in in a few days, I'll know by then and update if necessary.
11. Good thought, we'll update that!
12. Ah, very good point, those footprints should have their paste layer deleted. I'm also not holding my breath for this to work, it's honestly just a test. I have a feeling we'll end up deleting them down the road.
13. Interesting thought! We've noticed that sometimes U11 shuts off if there's ringing on the VIN line, which we're hoping to fix using a higher value decoupling capacitor on the input. At 24v input, we have no issues with the gate failing to trigger, or chip damage. Would adding a zener clamp help with the shutdown issue?

If you'd like to make some improvements, check out the [contributing page](https://github.com/index-machines/index/wiki/Contributing)! This board is just a beta, so probably a couple more spins before we have something ready for release.

ZOUDIO, 2022-05-17:
1) Pin-in-paste is used quite a lot nowadays. Did you use a stencil or applied manually? If manually, are you sure you added as much solder as a stencil would add?
Keep in mind that a lot more paste makes it in the hole than just the regular surface covering of a stencil, because its pushed in the hole with the squeegee.
2) I totally understand the module approach instead of embedding it. Was just questioning the benefit of supporting so much steppers.
It would reduce product complexity to choose a suitable stepper for the customer and just run with that.

And on this point: the jumpers seem to tie the stepper module pins to either a STM pin or 3.3V.
Isn't the same functionallity just possible by setting the pins of the STM high in software?

3) Almost all drivers in that footprint have the same pinout, so in that way it would be fine.
The only benefit to wiring the seperate pins is that instead of choosing to receive OR send, you can disable both.
4) Ah okay, interesting.

7) Sounds good. Less 230V danger, and in theory you could even remove the reverse polarity protection.
But it does not cost much, so you might aswell leave it on incase somebody uses a DIN with reversed pinout.

9) Did not look at the lay-out close enough and did not spot the 3V3 layer.
Most off the inefficiencies can be avoided by having a more rectangular lay-out, but if you stick with 100x100mm, this is quite good.

13) Yes I would recommend a 10V or 15V zener clamp to avoid problems. Look online to see how to implement it.

closed 2023-08-30 by G-Pereira