opulo-inc/lumenpnp #566: [Release assets] Position files are missing fidutial location

G-Pereira, 2022-06-06:
The release assets/artifacts that originate from our CI do not include fidutials on the gerber position files

elafontaine, 2022-10-19:
Hi @G-Pereira ,

I've started looking into this issue.

I must admit that I need your help into understanding this one :).  I'm trying to figure out what exactly you're looking for in those gerber file :).  For me, fidutials only means the "scale" into the file.  However, gerber files are made to be specific to certain metrics.  If you could help me identify what isn't shared in the gerber files, I could resolve the issue for you :).

Could you provide me the information in regards to what you are looking for exactly ? If you do, I'll do what I can to see it done ;).

G-Pereira, 2022-10-19:
Thanks @elafontaine . The problem is that the position files do not include fiducials (marks on the PCB used by the pick and place to calibrate reference position and rotation).

[image]

When loading position files (which describe the every component location on the PCB) these should include fiducials as well so it can be configured on the machine's software (for example OpenPnP)

elafontaine, 2022-10-20:
ok, I've found some example of this :) thank you.

Now, I'm trying to understand the "position file" you're talking about.  In the kibot configs, the position file output a csv file.  

I've found the issue you were talking about.  I'll read into the logs of the action that generated them and why kibot makes it that way.  I tried to run it on my kicad software, but it always generated the FID components on the file for me... my guess is that one of the option is removing it... got to figure out which one :D.  

Anyway, I'll continue on Friday.  For those trying to understand the issue, download the latest release "PCB" package from
https://github.com/opulo-inc/lumenpnp/releases .  From there, look as the *_cpl_jlc.csv (or any "cpl" file (Component Placement)).  These file are missing the FID x (where x is an integer) which would be the position of the Fiducials.

You can then open the component you looked at in KiCad and "File" > "Fabrication Output" > "Component Placement (.pos)"
In there, change the output format to csv and observe the difference :D.

Stay tuned

G-Pereira, 2022-10-20:
Thank you!
Just one minor thing: `mobo_cpl_jlc.csv` and `mobo_cpl_jlc_conn.csv` should not have the fiducials.
The ones that need to have it are `mobo-bottom_pos.csv` and `mobo-top_pos.csv`

elafontaine, 2022-10-24:
Ok, started to look tonight.  

Kibot default exclude filter on the "base_filter' remove the FID; 
- Src ; 
       - https://github.com/INTI-CMNB/KiBot/blob/master/kibot/fil_base.py#L217
       - https://github.com/INTI-CMNB/KiBot/blob/master/kibot/fil_base.py#L17
The configuration of "global: variant: default" in the config is what seems to be what causes the issue.

My current hypothesis is that the global setting apply the "default" filter which include the _mechanical one.  I'm trying to understand why we set it like that and why we couldn't just remove that from the setting (and apply the "default" where it is needed and/or specific filter where we need something specific).  

To be continued.

Question; Would we mind keeping the output of the generation of files at CI time for releases only.  (Food for thoughts ?) 

Output on my ubuntu server using kibot as CI does; 
```
- 'Pick and Place Location File' (pick_and_place_file) [position]
DEBUG:Output destination: /root/lumenpnp/pnp/pcb/mobo/mobo/gerbers (kibot - kiplot.py:361)
DEBUG:Filters reset (kibot - fil_base.py:161)
DEBUG:Applying filter `_mechanical` to exclude (kibot - fil_base.py:154)
DEBUG:Excluding 'FID1': Field 'reference' (FID1) matched 're.compile('^FID', re.IGNORECASE)' (kibot - fil_generic.py:160)
DEBUG:Excluding 'FID2': Field 'reference' (FID2) matched 're.compile('^FID', re.IGNORECASE)' (kibot - fil_generic.py:160)
DEBUG:Excluding 'FID3': Field 'reference' (FID3) matched 're.compile('^FID', re.IGNORECASE)' (kibot - fil_generic.py:160)
DEBUG:Excluding 'FID4': Field 'reference' (FID4) matched 're.compile('^FID', re.IGNORECASE)' (kibot - fil_generic.py:160)
DEBUG:Excluding 'FID5': Field 'reference' (FID5) matched 're.compile('^FID', re.IGNORECASE)' (kibot - fil_generic.py:160)
DEBUG:Excluding 'FID6': Field 'reference' (FID6) matched 're.compile('^FID', re.IGNORECASE)' (kibot - fil_generic.py:160)
```
When I remove the "global: variant: default" it does not do this.

G-Pereira, 2022-10-24:
That makes sense. I would just check if that wasn't put there to trigger something else that was missing before and would change to the actual default value as you said, I wouldn't remove it. If I recall correctly @daveismith we had the approach of specifying all parameters even if they were set to their defaults for transparency and consistency concerns, correct?
We did that to avoid messing our things up if KiBot decided to change their default values at some point.

daveismith, 2022-10-24:
My recollection is that we went with the kibom type as it was producing the desired output format. The only filtering we were intentionally doing was to covert names / packages for JLCPCB.

elafontaine, 2022-10-25:
I'll push the PR tomorrow during the day, but I ran a diff of the files between applying the variant option individually on any tasks vs using the global setting;
- The differences are all timestamps in the files for text files (the only exception being the html file having some ids being different)
- The binaries are all differents, but I can't know what is.... my guess is that there is also some timestamp in them.

Over-all, it doesn't seem to affect the content.  I wouldn't be able to confirm the binary file content, any leads I could follow on?

elafontaine, 2022-10-27:
If I understood correctly, all you wanted is the default behaviour of kibot for position file (gerber folder).   This could be easy if we remove any "variant" to be applied on it.  We can have other variant in other tasks (like the global Kibom one). 

Please have a look and let me know how you would wish to proceed.

elafontaine, 2022-10-27:
By the way, I realised I took an assumption that this was needed on every config file (for the position file).  Was that wrong ?

G-Pereira, 2022-10-27:
Please share a link to the workflow run where those files are being generated and the `<board>-top_pos.csv` include the FID entries.

elafontaine, 2022-10-27:
I'm not sure about what you're asking.  You want me to send the command I ran on my ubuntu server ? 
or just the file that shows the csv has the FID in it ? or zip the whole gerber folder and share it ?   Or do I have a way to trigger the CI ?

elafontaine, 2022-10-27:
Ok, I think I found what you meant (I didn't knew that github runners were free, nor that I could just activate a "workflow").

https://github.com/elafontaine/lumenpnp/actions/runs/3338496971/jobs/5526227581