opulo-inc/lumenpnp #572: Mobo needs protection from shorts due to dropped components

InfiniteNesLives, 2022-06-06:
I can't recall who brought up this issue in Discord awhile back, but it's a good one.

There is a real risk that dropped PNP components could fall through the staging plate and short out pins on the mobo causing irreversible damage.  This would be especially painful in the middle of a PNP job.

I'm going to cut out some cardboard quick to temporarily shield my mobo from this risk.  But some sort of FDM shield/enclosure should be added to the default design and included in the assembly docs to protect users from this risk.

A basic box that also acts as mobo stand-off for staging plate mounting could work well and remain completely open on the downward/bench top facing side with all the connectors.

stuartpittaway, 2022-06-07:
Alternatively, the staging plate could be made to not have the holes directly over the mobo area.

atanisoft, 2022-06-07:
I proposed some FDM plugs for the staging plate but that may end up blocking an area from having a strip/tray feeder (also a good option to cover the MB)

InfiniteNesLives, 2022-06-07:
Yeah could take a number of different solution paths.  But at a minimum would probably be good to make this risk more visible to users.

The mounting holes could remain useful especially now that the staging plate is non-magnetic fiberglass for opulo kits.  Can't really remove the holes from existing users or current opulo staging plate inventory. 

My current temporary fix is pretty ugly, but should work for now.  Dont want to disturb my rail of feeders right now.  Next time I need access to something and remove all my feeders I'll probably implement a more proper fix. 

[image][image]

[image]

Magpie-81, 2022-09-08:
I made a printable protection plate for my machine. The files are available on Printables: https://www.printables.com/model/242429-lumenpnp-mobo-protection-plate

closed 2022-09-16 by sphawes