opulo-inc/lumenpnp #578: MOBO Rev 4 schematic and PCB design Improvments

rfm24, 2022-06-19:
Hello There Stephen, this project you're designing is great, and came a long way! Congrats.

Looking at your board I found a few things that can be improved on your schematic and PCB layout, and can/will change your layout if you apply them.
Of course I'm just sharing this with you because I'm an enthusiast and like PCB designs, you're not obligated to any of what I wrote here. I've been using KiCad, maybe for 3 months now, I was using proteus but found KiCad more intuitive and easy to work with for PCB design than Proteus (the version I was using either way). And being free also helped...

Schematic Improvements:
1 - Add Labels to all nets so you or anyone else can easily identify them in PCBnew.

2 - Add notes to your schematic such as filter cut-off frequencies, max power consumption, signal path from the USB connector, app notes, etc.

PCBnew Improvements:

0 - For MLCC caps, Resistors that won't drive much current, with power ratings of 1/10W, 1/8W, assign smaller footprints, like 0402 or 0603, those can easily be found in packs in places such as LCSC, etc, they sell 100s of them for maybe less than 1$ ?
 
1 - What's the reason behind having many ground planes, you could have just had 2 GND planes on both internal layers
that would actually be better than having 3 GND layers, specially because a few chips like the MCU can have pretty fast signals
this will end up causing problems such as crosstalk and EMI, Signal Integrity, GND planes on the signal layer make the signal from one trace coupple into other traces. So the trace that should behave as a microstrip line, will now behave as a coplanar waveguide do to the GND plane close to it, this will add capacitance and this capacitance will affect the characteristic impedance of the trace
	Since you have multiple voltages running through the board wouldn't it be better to have higher layer count ?

	Also review your power design techniques, you should use copper fills/pours on your design for Power Delivery, to minimize
inductance and noises, furthermore separate your Switching PS from your linear regulator, Inductors generate a lot of noises, that can coupple to your not so noisy LDO but use power puddles instead of traces to connect each component and be considerate about "return paths", you should always try to use a solid ground plane as the return path.

2 -  Every time we change signal layer with a via, we still need to consider the "return path ", so for this
you should have a "return via" (it's a via that shorts the GND planes) connected to GND close to that SIG via
use your fab website to determine what capabilities they offer for hole spacing. 
Still in this topic of vias, for " power vias ", like vias used in decoupling caps, keep them as close as possible to the pad, but not in the pad! this will reduce the inductance but that might not be enough, for decaps which use VCC and GND vias, you should keep them as close as possible to minimize inductance as well. In your design there's some inconsistencies, a few vias that are indeed close to the respective pad, but there are others that are not close enough.

3 - For USB circuitry, connections and Power Delivery, I recommend you to look at ST App Note: AN4879I  and App Note: TA0357 for STM32 MCUs, both can be found in STM website
specially the resistor for D+ line, and PWD over USB, it's also a good practice to use low capacitance ESD protection i.e. TVS diodes, do to the high bandwidth of USB connection according to the speed you're using, also it is optional but maybe add a Common-Mode-Choke for EMI/EMC filtering ?, USB ports usually are very noisy...

On the topic of ESD protection, I also recommend you to use TVS diodes on you SWD connector or any physical connector, to protect them against accidental human touch, some ICs, OPAMPs, MCUs, etc. These devices are ESD sensitive, MOSFETs for example, they can short if not correctly handled.
For ESD devices, you should take in consideration the following parameters:
 a) Low capacitance, higher capacitance will affect rise and fall times and the speeds at which the system can run.
 b) Bidirectional diode, instead of unidirectional.
 c) Dissipated Power, the higher the better

4 - I'd also recommend separating your analog circuit from any digital circuit because of the fast switching signals from digital circuits this of course will increase the overall size, but I'd go for that route, to minimize artefacts such as crosstalk, EMI/EMC, Impedance discontinuities, etc. So keep them apart is what I recommend, specially for mixed signal boards

5 - Use smaller sizes for decoupling caps for MCUs, such as 0402, or 0603 max, to maximize spacing and positioning, and also minimize loop inductances, caps with longer leads or bigger sizes have higher loop inductances than smaller ones, although they might have higher ESR(place as close as physically possible to the power pins of the IC))

6 - Separate fast switching SIG from lower switching, such as USB, SDA, SPI, as a rule of thumb, typically they should have a spacing of at least 3x the trace width away from each other, keeping them close will affect your trace's characteristic impedance, changing it's value, creating further problems with signal integrity, predominantly because as mentioned in point 1, you will get a lot of crosstalk, so using the technique mentioned will improve crosstalk.

I recommend you to watch these videos on youtube, some of these are 2 hours long, but I really recommend you to watch for better design techniques
- "Altium Academy - You must Unlearn what You have Learned - Eric Bogatin"
- "[LIVE] How to Achieve Proper Grounding - Rick Hartley - Expert Live Training (US)"
- "How To Improve Your PCB Designs (Common Mistakes) - Phil's Lab #18" from phil's lab youtube channel.

Magpie-81, 2022-06-29:
I agree with a lot of points you made.
As a reference, i am a professional electronics engineer for industrial applications doing PCB and schematics designs for boards that have to pass EMC testing. (more than 20y experience in the industry)
This PCB design here would unfortunately not have passed my review, before even considering to spend some time for a precompliance test at our in house lab...
A few critical points that i've seen just having a quick look at the design:
- Not enough decoupling capacitors at the interfaces. This is especially critical on the USB interface side, but other interfaces like the feeder connection or even the power input stage suffer from this as well. Add more capacitors directly at all the connectors (10nF to 100nF).
- Some decoupling capacitors are badly placed and layouted. This is especially critical on the main controller, but the USB hub is another critical part that needs better care with power decoupling. The path should be from plane to via to decoupling pad to IC pad, and as aready mentioned, don't forget the rest of the loop over the GND pads.
As an example, R2, the I2C_SDA resistor is way less critical than the decoupling capacitors next to it, so the capacitors C6 and C12 need to take that space close to the ICs power pins.
- For an other extreme example, look at the +5VP power that goes all around the board from the USB side to the power converter. What ever RF noise gets picked up at the cameras and it's cables has plenty of opportunity to get coupled into other parts of your circuit, like the pressure sensors, microcontroller and the USB hub.

- Sometimes you even managed to squeeze a via almost into a component pad. This can be a big problem at the reflow stage when the via "sucks" away the melting solder paste from the pad. This is a reliability and production issue. Without good inspection, an electrical test might not see it as an issues immediately, if it's on a decoupling cap. But the device might have more issues in the field and could fail EMC testing.

- Another big problem is the cutting of the internal 3V3 plane in the center of the PCB with long vertical tracks. Sure, you can use the internal power plane for some signals, especially where it's tight. But you need to leave enough room in between for the plane copper. Try to use tracks as long as needed but as short as possible when cutting thru the internal planes. And keep an eye out for where and how you cut into a plane.

- I agree with rfm24 mentioning transorb diodes to protect interfaces. Maybe you get away without them, but having a footprint in place and a component that costs a few cents is always better than having a board getting "zapped and killed". One interface i see highly critical here (apart from the USB that is always risky) is the feeder interface, as that is probably something that gets a lot of handling and plugging stuff in/out in normal operation when the user needs to set up a new job or replace reels.
On the power delivery side, in my designs i tend to isolate regulators always with a series element in the power path. This can just be a 0 ohms resistor, but usually a ferrite bead (with decoupling capacitors) will be used to help get rid some some emissions or incoming interference. Sometimes these elements are handy to remove in repair jobs as well, as you can isolate different power sections to make it easier to find and replace a faulty component.

- It's basically already standard best practice for me to add capacitors to GND on the I2C SDA and SCL lines (usually around 100pF). Hard to get any device thru RF radiated immunity testing without these capacitors. At some interference frequency the bus will just lock up and not communicate anymore.
-  Same with reset signals, they need capacitors to GND directly next to the input pins of the IC, or every little ESD event near the device will just trigger a reset of the device.

- Another thing i did find somewhat funny is that you tried to do a length tuning on the differential pairs (USB / RS485). I really don't think this is needed or helps in any way here. It would be way more important to try and tune the impedance to the required specification with proper track width and spacing, and keeping the 3 hub output pairs away from eachother as much as possible to avoid crosscoupling. Without knowing the boards layer stackup (and define that to some degree) the impedances are really a "hit and miss" target here tho. Especially when you use the cheapest overseas pcb fabs. They just throw what ever core and prepreg they find into the board to make a multilayer stackup work. 
And as rfm24 mentioned, some extra filtering and protection at the USB connectors is worth considering.

There is definately a lot more that needs to be checked and lots that can still be improved. I did not do a full review of the design, just had a quick look into the files to "know what to expect" as we are starting to assemble our V2 Opulo Kit.

If you would like to get a better review of the schematics and pcb before you release the next board in to the wild, i might dedicate some time into it...

theacodes, 2022-06-29:
Hey folks, thanks for the critical feedback. We definitely want to keep improving all aspects of the machine. That said, I think at this point it's clear the PCB design needs more eyes and more eyes with more experience. I don't think it'll be constructive for folks to add any additional critiques at this point.

If you'd like to get more directly involved, I recommend joining our [discord](https://discordapp.com/invite/TCwy6De). I would definitely recommend coordinating with the devs before doing any major work just so we make sure we're not stepping on each other's toes.

troth530, 2022-07-30:
The comments are well thought out, constructive and informative.  It's awesome that people are willing to take time to help improve it.  If this were buried in a discord, the community would lose out on some important design principles.