opulo-inc/lumenpnp #580: Documentation for PCB Staging Plate

Memnarch1113, 2022-06-20:
Slightly different documentation required for step 8 [here](https://docs.opulo.io/docs/mechanical-assembly/mount-y-to-frame/), and for [bottom camera mount steps ](https://docs.opulo.io/docs/mechanical-assembly/populating-staging-plate/#bottom-camera-mount)which will need to include the foot.

theacodes, 2022-06-20:
Hi @Memnarch1113, could you let me know what needs to change? As far as I know the PCB staging plate shouldn't make a huge difference to assembly.

Memnarch1113, 2022-06-21:
yeah, what @atanisoft said. I'm realizing that this needs to be in the docs repo, so I'm closing this in favor of that one sorry about that! https://github.com/opulo-inc/docs/issues/1