opulo-inc/lumenpnp #590: Marlin Firmware Links Broken Due to Index -> Opulo Name Change

feldon30, 2022-08-03:
The ["Update the Firmware" docs](https://docs.opulo.io/docs/motherboard/update-firmware/) point to:

- Marlin Bugfix 2.0.x firmware
- Configuration.h and Configuration_adv.h for bugfix 2.0.x.

However Index/Rev_03 is no longer the correct path for these config files. It must be corrected to "[Opulo/Lumen_REV3](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Opulo/Lumen_REV3)".

NOTE: If someone wants to flash Marlin Release 2.1, then they must get the config files from "Index/REV_03" and some modification of platformio.ini is required to get it to work.

atanisoft, 2022-08-03:
https://github.com/opulo-inc/docs/pull/26 covers this likely.

As for building 2.1, I did not have to modify anything in platformio.ini to get a successful build. I also did not use the auto-build plugin but instead used commandline to build.

feldon30, 2022-08-03:
Awesome re: the fix by Yandrick.

Thanks for the info on the 2.1 build. Probably user error then.

jonnerd154, 2022-08-04:
Came here to report this same issue, but seeing this conversation lead to a resolution of my problem.

@feldon30 I had trouble building for a while, too - after experiencing the stale link confusion herein, I realized I was still trying to use the (old) 2.0.x firmware with the (newly found) 2.1.x configuration. Yandrik's opulo-inc/docs#26 addresses both. Maybe you were doing the same?

Firmware here: https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.1.x.zip
Configs as noted in opulo-inc/docs#26

Yandrik, 2022-08-04:
That's actually a problem I had as well. I'll put a note about it into the PR to make this less confusing, hopefully

jonnerd154, 2022-08-04:
Awesome, thanks! 

Were you able to flash from inside Marlin Auto Build, or did you fall back to using dfu-util on the CLI? I gave up and used CLI.

Also, I had quite the struggle with drivers for DFU: had to download the STMicro driver in Windows Update, then switch it to WinUSB with Zadig after...messy! 

Was that just me?

atanisoft, 2022-08-04:
PIO and dfu-util don't seem to play very nicely for most people. Even on Linux I abandoned that option and instead used dfu-util directly, and that was with PIO CLI mode (no auto-build stuff).

feldon30, 2022-08-04:
PlatformIO and people building Marlin seems the HARDEST way to update the firmware for the reasons stated above.

I think focusing on the binary and dfu-util is going to bring simplicity and repeatability for users.

Yandrik, 2022-08-04:
Luckily, I've never had problems with ABM until now, at least for the building part. The auto flashing I never even tried. `dfu-util` worked well for us tho, and did its job. Unfortunately, it doesn't reset the chip automatically, so manual power-cycling is still necessary to get the machine out of DFU mode.

I'll rewrite that part of the documentation in the PR, and add a note that PIO flashing might work, but doesn't for many people.

Yandrik, 2022-08-04:
Rewrote flashing to using `dfu-util`. I'd appreciate it if you guys could check whether the new flashing docs is easy to understand and such.

jonnerd154, 2022-08-05:
Makes sense to me, thanks @Yandrik. One suggestion:

@CurlyTaleGames [suggests a slightly different call](https://discord.com/channels/688723670461448225/885309109929406474/993721249006100530) to `dfu-util`, and this command *does* make the controller exit DFU after an update.

```
dfu-util -d 0x0483:0xdf11 -a 0 -s 0x08000000:leave -D .pio/build/Opulo_Lumen_REV3/firmware.bin
```

`-s 0x08000000:leave` is the critical bit: it uses DfuSe to request the device "leave" DFU. Man page [here](http://dfu-util.sourceforge.net/dfu-util.1.html) for reference

So you could update the command and remove the need for manual reset in step 11. Otherwise it all looks good to me! Thanks for doing this.

Yandrik, 2022-08-05:
Sounds awesome!
New command is pushed (https://github.com/opulo-inc/docs/pull/26/commits/03cc66c7ce5c43a365f555c511c2da9c2ed058d1), and should now be merged into the official docs  :tada: 
Thanks for the tip!

closed 2023-08-30 by G-Pereira