opulo-inc/lumenpnp #598: Right Front Leg Limit Switch Recessed Slots in Wrong Place

solarmax1, 2022-08-26:
Mounting a limit switch on the right front leg is not a standard configuration, but if one may wish to add the limit switch to support independent Y motor drivers.

The limit switch needs to be mounted with the connector pointing upward, and the switch facing the back of the machine. The two small mounting holes are in the correct place, but the recesses for the connector and switch do not line up with the limit switch board.  These recesses need to be swapped positions and moved.

lucian151, 2022-09-07:
This issue likely stems from the fact that front-right-leg is modeled by mirroring the front-left-leg. We will have update the recessions on the front-right-leg manually. We could also consider adding recessions to the front-left-leg that are correct for the right side leg when mirrored.

G-Pereira, 2023-08-30:
Changed to sensorless homing, so this is no longer applicable