opulo-inc/lumenpnp #601: Mobo Rev4 Decoupling Capacitor Issues

Magpie-81, 2022-09-08:
Input capacitors C25 and C38 are specified as 22uF/25V/0805 capacitors. The nominal supply voltage here is 24V. These need to be specified for at least 35V to have some headroom for input voltage fluctuations.
The other issue with running them this close to the spec voltage is DC bias capacitance degradation. They will not provide the required filtering because they are never actually 2x 22uF at this level of voltage.
Since 22uF/35V ceramic capacitors are a pretty rare breed, even in larger than 0805 packages, this would rather require a tantalum capacitor and some lower capacitance ceramics in parallel to provide the required filtering.
The other option would be to lower them to 10uF/35V ceramics, at least there are enough options on the market there.

It is always a good idea to add a smaller capacitor (10nF...100nF) in parallel to bulk capacitors (>=10uF) to provide a wider filter bandwith against higher frequency signals (ESD and RF noise in particuar).

There should be a filter capacitor (10nF...100nF) directly placed at the power input connector as well.

U7 (servo driver gate buffer IC) does not have a decoupling capacitor.

USB power needs a decoupling capacitor directly at the USB connector. (best option here would be a bulk capacitor and a small capacitor in parallel again). This will massively improve the performance of the ESD protection diodes already in place there, but is required to filter out any noise as well.

Generally it is best practice to add decoupling capacitors to all the interfaces directly at the connectors. A prime example are the two Light connectors where the voltage supply track runs around half the board from the connector to the next capacitor near the voltage source.
The wiring for the light will run from the mobo all the way to the placement head, along with the motor drive wires, so it is very likely that there is significant noise coupling into the 5V supply rail that would then affect basically all other 5V devices connected to other interfaces on the mobo.

zakrent, 2023-03-10:
We've experienced a failure of motherboard caused by those capacitors. VIN got shorted to ground. C25 and C38 capacitors according to the LCSC SPN are rated only for 6.3 V (C5674) and I'm suspecting that that's what we had in our machine. After removing those capacitors the short is gone. I'm expecting that there are going to be more failures caused by it in different machines.

atanisoft, 2023-03-10:
C2012X5R1V226M125AC might work as a drop in replacement for C25 and C38, it's rated for 35v but is around 3x the cost to a 25v version. C3216X5R1V226M160AC might also work but is a 1206 package.

G-Pereira, 2023-08-30:
Capacitor selection revised in next Mobo version