opulo-inc/lumenpnp #608: No x y axis limit switches LumenPnP ver 3

hedon77, 2022-09-19:
Lack of space for limit switches on the x and y axes makes 
it difficult to use the mobo ver.3 in the new LumenPnP design.

nkappler, 2022-10-02:
The Lumen now uses sensorless homing by driving the axis into the hard stop and measuring the current increase in the stepper motors. No Limit switches required.

jonnerd154, 2022-10-12:
This can be accomplished with the V3 mobo. There are two jumper wires required, but after those modifications and a firmware update, you'll have the same behavior as a V4 mobo, with one exception: the Y-axis won't auto-tram, as there is only one Y axis motor driver shared between the two motors.

1. Disconnect your X and Y limit switches from the mobo and remove them from the machine.

2. Ensure both axis' can move all the way to the homing position, against their endstops without being impeded. Check that the ends of X-gantry belts don't prevent the Y-gantry from contacting its endstop.
 
3. Run jumper wires like this - from the diag pin on each driver, to PIN 1 on their respective limit switch connectors:
[image]

4. Setup a Marlin build environment using [these instructions](https://docs.opulo.io/docs/motherboard/update-firmware/).

5. Adjust Configuration.h and Configuration_adv.h as below. You'll be modifying existing lines, or uncommenting them. All of these values should already exist in the default configs!

#### Configuration.h
```
#define X_MIN_ENDSTOP_INVERTING false
#define Y_MIN_ENDSTOP_INVERTING false
#define ENDSTOP_INTERRUPTS_FEATURE 
```

#### Configuration_adv.h
```
#define SENSORLESS_BACKOFF_MM { 2, 2, 2, 2, 2, 2 }
#define HOMING_BUMP_MM { 0, 0, 2, 0, 0, 0 }
#define HOMING_BACKOFF_POST_MM { 1, 1, 0, 0, 0, 0 }

#define SENSORLESS_HOMING
#define X_STALL_SENSITIVITY 68 // You may need to adjust this!
#define Y_STALL_SENSITIVITY 58 // You may need to adjust this!

#define IMPROVE_HOMING_RELIABILITY
```

**Note:** you can/should adjust `*_STALL_SENSITIVITY` and `*_CURRENT` in `Configuration_adv.h` to optimize for your machine. I use 32-step microstepping and linear rails, and the values above are what I use, including keeping the default`*_CURRENT_HOME` being equal to `*_CURRENT`.

6. Build the project.

7. Power the machine, connect USB.

8. Jump the mobo to bootloader mode:
  - Press and hold the "BOOT" tactile button
  - While holding BOOT, press the RESET button for 1 second and release
  - Wait 1 second
  - Release the BOOT button
  - The mobo should appear as `STM32 BOOTLOADER` in Device Manager-->Universal Serial Bus devices

9. Flash the board. There's a good chance the Platform.io "Flash" button won't work, so just use this command in the command console:
```
~/.platformio/packages/tool-dfuutil/bin/dfu-util -d 0x0483:0xdf11 -a 0 -s 0x08000000:leave -D .pio/build/Opulo_Lumen_REV3/firmware.bin
```

10. Run homing, and be ready to cut power if things go poorly. If one of the axis triggers the bump-stop before it's moved all the way to the physical endstop, decrease that axis' `*_STALL_SENSITIVITY`. If it contacts the endstop and grinds the belts, increase that axis' `*_STALL_SENSITIVITY`. It will take some tuning, but worth it.

**NOTE:** You can use [M914](https://marlinfw.org/docs/gcode/M914.html) to adjust the bump sensitivity, and [M906](https://marlinfw.org/docs/gcode/M906.html) to adjust the current at run-time. Once you find a combination you like, then change those values in the firmware, build, and flash. Saves lots of time.

sphawes, 2023-05-15:
thank you for the note @hedon77! the v2 -> v3 change was intended to remove the mounting for limit switches, given that the motherboard in the v3 machine has sensorless homing, and the space was needed to add a cable chain. If you have a REV03 motherboard, I recommend keeping the prints that have the limit switch mounting holes.

Thank you @jonnerd154 for the super detailed write up!!

closed 2023-05-15 by sphawes