opulo-inc/lumenpnp #614: Marlin: Stepper motor current setting for I & J (Z rotation) axes is too high

solarmax1, 2022-10-08:
The NEMA11 hollow shaft motors have a rated current of 800ma.  
[https://www.robotdigg.com/product/798/NEMA11-hollow-shaft-stepper-for-Pick-and-Place-Machine]

The Opulo-provided Configuraton_adv.h for Marlin has the I & J axes current defined as 700ma.  This value should be no more than: 800ma * .707 = 566ma.  At the 700ma setting, the motors get very hot when running the OpenPnP up looking camera calibration which runs for about 20 minutes and has the head rotating every time the head moves, which is almost constantly.  The motors can also get hot when when holding position.  I've tested the 566ma setting and I don't see any difference in performance, and the motors run much cooler. 

The solution is to reduce the current to 566ma by changing #define I_CURRENT and #define J_CURRENT to 566 in the configuration_adv.h file for Marlin.

stuartpittaway, 2022-10-08:
@sphawes aren't you changing the marlin configuration files at the moment? Can this be dropped in as well?

sphawes, 2022-12-05:
hey @solarmax1 thank you so much for the note! I've run some tests for this issue, and havent had an issue with the motors getting too hot at hold current (which we have set to be half of the max), at steady-state they only get to 27.6 degrees C with current set to 700ma. That being said, they're not being driven at full current during something like camera cal as you mentioned, so I totally agree that we should reduce the current!

Is your machine from Opulo, or self built? What is the output of `M122` from your machine? I just want to make sure there isn't something else going on with your machine, as I've never had the motors get too hot to touch.

Have you ever had issues with the rotation motors skipping steps with 566ma? My inclination to keep the driver current high here is to make sure we don't get skipped steps when rotating, and causing misplaced parts. I ran some tests here with your recommended current setting, and it seems to be plenty of torque, but i was wondering if you've seen any regression with your settings.

Thanks again, much appreciated!!

solarmax1, 2022-12-05:
Hi Stephen,

My machine is self-built.  It is interesting that you cannot get the motors to heat up. I'm using the standard RobotDigg.com hollow-shaft motors.  A couple others are seeing the same issue on Discord. I don't know if their machines are self-built or kits.  Mine were getting hot enough that I didn't want to touch them, but I didn't measure the temperature.  The person I responded to on Discord had the plastic mount softening.  

I don't think it takes much torque to spin a part around with nothing touching it.  My motor is not set real fast because I wouldn't want for any large parts to either slip on the nozzle or fly off.  I have not seen any skipped steps issues.  The torque is pretty strong if I try to move it when being held in place.     

Attached are the results of M122.  I'm running 2 nozzles, dual Y axis drivers, and separate sensorless homing on each Y axes.  The output will look a little different than what you are used to.   Let me know if you have any other quetions.

[M122.txt](https://github.com/opulo-inc/lumenpnp/files/10157569/M122.txt)

mwoodruff, 2022-12-06:
For what it's worth, I just put together a kit and both A/B motors are running super hot. Hot enough they melted the carriers, which I reprinted in ASA. My M122 shows no connection to the A/B motors, and they do not respond to M906 settings. The X/Y/Z motors do respond to the current changes. Perhaps there is something else happening for me as well.

solarmax1, 2023-02-25:
Others on Discord with the same problem:
https://discord.com/channels/688723670461448225/739250421893300264/1077578954505986088
https://discord.com/channels/688723670461448225/739250421893300264/1076493825322799184

Workaround (instead of recompiling Marlin) is to use the M906 command in the gcode driver “device connect” command to set lower current for the rotational steppers.  This should be added to the Lumen OpenPnP setup documentation.

sphawes, 2024-02-19:
we now have lowered L and R stepper motor currents baked into the most recent firmware release, along with openpnp configs. for folks that want to just make a quick change for their motor currents, this page on the docs site will walk you through it: https://docs.opulo.io/guides/motor-current/

closed 2024-02-19 by sphawes