opulo-inc/lumenpnp #620: FTP POS file missing

gabeseltzer, 2022-10-25:
The FTP POS file is missing from the latest release 3.0.1! [See docs here](https://docs.opulo.io/openpnp/ftp/setting-up-the-board/#finding-the-ftp-board-location)

elafontaine, 2022-10-27:
Maybe I could look at this ... I got some time and just went into the CI part.

elafontaine, 2022-10-27:
I'm not observing the problem that was reported.  I think I'm missing some information ;
- Both the `3.0.0` and `3.0.1` release contain a `ftp.zip` which in turn contain a file called `ftp-top_pos.csv`
- I've looked at the file history of the kibot configuration for FTP and the export-ecad github task, but not much changed.  This indicate to me that it was never generated as a "ftp-top.pos" file (not to confuse with the CSV output).

I could add a kibot task to generate the file, but is my hypothesis wrong ?  Do we have a precedent when it worked ? 

If not, should I go ahead and generate the .pos file ?

elafontaine, 2022-10-31:
I've checked again this evening.  I see the description changed and the text removed any reference to the file as if it was available.  Please give me context as if this is resolved or not.  I'm open to either ; add  the file or change the description (in a way to show how to export the file for the job.

G-Pereira, 2023-09-14:
The FTP board is not meant to be used for placing components so the POS file is not useful for machine calibration, which is this board's intended use.

closed 2023-09-14 by G-Pereira