opulo-inc/lumenpnp #627: Add clearance for access to bolt heads on Y gantries

CurlyTaleGames, 2022-11-07:
This is more of an improvement than an issue, but it'd be nice to be able to access these screw heads on the Y gantries, should be a pretty simple cad change.

[image]

lucian151, 2022-11-14:
Hey @CurlyTaleGames this is already fixed in the V3.0.1 file set and onwards -are you seeing this issue with the most recent design release?

CurlyTaleGames, 2022-11-14:
@lucian151 weird, I downloaded V3.0.1 from the releases and printed everything:
[image]

I just checked out y-gantry-x2.stl and it matches the file I printed:
[image]

Here is are screenshots of the x-idler-mount.stl:
[image]

And the x-motor-mount.stl:
[image]

I just downloaded V3.0.2 from releases and those parts match the parts I have in V3.0.1

If those cad changes have been made it doesn't look like they are in the STL files in the releases

CurlyTaleGames, 2022-11-20:
For fun I downloaded the Freecad x-idler-mount and exported it out as a stl 
[image]

Then added that stl as well as the stl from the 3.0.2 release into prusa slicer and matched them up, and can confirm they are the same:
[image]

Is there a CAD branch that has this change that needs to get merged in?

lucian151, 2023-01-04:
@CurlyTaleGames Ah I see now, I'm sorry for misunderstanding your suggestion. I had updated these pieces in 3.0.1 to fix bolt heads that were sticking out far enough to interfere with the removable x-gantry assembly. What you'r suggesting is entirely different than my earlier change, and I'm not sure how I didn't realize that earlier.

This isn't a bad idea - I'll take it into consideration the next time I have to touch these two models. If you're interested in taking a stab at it before I get to it, I'd be happy to review a PR!

closed 2023-09-06 by lucian151