opulo-inc/lumenpnp #630: Vacuum pump mount - can't replace rubber bands without taking apart

CurlyTaleGames, 2022-11-08:
[image]

The vacuum pump mount has notches for the rubber bands to sit inside of, but when you screw the mount to the build plate you can't replace the rubber bands when they go bad without unscrewing the mount. 

If there was a little clearance between the mount build plate it'd be much easier to do.

atanisoft, 2022-11-08:
When rebuilding my machine over the weekend I found the original kit supplied rubber bands had snapped and was faced with replacement (in-place). I faced a similar issue as you have reported here.

I'm thinking the section of the print to the right of the notch (in your photo) could be ~2mm taller to allow the notch to be slightly elevated off the staging plate, this would allow replacement of the bands without dismantling the entire assembly.

CurlyTaleGames, 2022-11-08:
I reprinted mine without the notch using negative volumes in Prusa Slicer, and the rubber bands held, but they aren't on as securely as I'd like. There definitely still needs to be a notch on the bottom, but a smaller one should be fine. Or maybe just make print taller on the side that mounts to the build plate

[image]