opulo-inc/lumenpnp #631: Ring Light Level Shifter Vih is Marginal

solarmax1, 2022-11-24:
The 74AHCT1G32 is used as a level shifter from 3.3V to 5V used on the ring light board.  VCC on the part is connected to 5V and the input level is as high as 3.3V for a logic 1.  

The Vih spec for the input when running VCC at 5V is  (.7 * VCC) = 3.5V minimum. This is below the 3.3V input signal level.  

This may work just fine, but just be aware that if people start seeing flakey issues with the LED's, this could be the problem. 

One example of a part that is pin compatible is the SN74LV1T34, which has a Vih of 2.02V and is designed for this purpose.

microfoundry, 2022-12-11:
Another option would be a true dual voltage level shifter such as TI's: SN74LXC2T45DCKR dual channel IC. Since it requires 3.3v & 5V it would be better suited to be place on the Mobo...

theacodes, 2022-12-11:
There's no need for a bidirectional logic level translator. This ubiquitous and inexpensive [74HC2G34 / 74HCT2G34](https://assets.nexperia.com/documents/data-sheet/74HC_HCT2G34.pdf) buffer is my go to for 3.3V -> 5V translation.

theacodes, 2022-12-11:
Upon further research, the specific part used by the ring light- TI's [SN74AHCT1G32](https://datasheet.lcsc.com/lcsc/1809251713_Texas-Instruments-SN74AHCT1G32DBVR_C43450.pdf) is explicitly rated for 3.3V -> 5V logic translation.

closed 2022-12-11 by theacodes