opulo-inc/lumenpnp #656: V3.02 Camera focus

ad6fp, 2023-02-20:
I had great difficulty getting the cameras to focus.  I did print the suggested 3d tool which was some help on the top camera.
However the access to the bottom camera is so tight that the tool did not provide much benefit.
I wound up disassembling the bottom camera and found that the focus lock screw had been over torqued causing the threads
to be munged.  A careful thread repair with an exacto blade and some very small amount of lubricant completely solved the focusing difficulty.   

Please inform the assembly house to not over tighten the camera focus lock screw.

lucian151, 2023-03-01:
Hello @ad6fp , thank you for reporting this issue and apologies for your difficulties. This feedback has been shared by you and a few others recently, so we've done a few things to address this moving forward:

- Reached out to ELP to discuss the issue and awaiting further feedback
- Introduce a bottom-camera focus calibration routine to the LumenPnP assembly line here at Opulo HQ, you can read this for yourself [here](https://ohai.opulo.io/lumen/staging-plate/#focus-the-bottom-camera).
- Created a knowledge card so our services team is informed on how to better advise affected users

I'm going to leave this ticket open for now, but please know that we've taken steps to stop difficult-to-adjust cameras from reaching Opulo customers moving forward. Hobby builders who have independently produced their LumenPnP may benefit from viewing the work instructions I've shared above.

>

JudgeBeeb, 2023-03-13:
Hi @lucian151. Are the 3D files for the jigs referred to in the above article available for download?

closed 2023-09-06 by lucian151