opulo-inc/lumenpnp #659: Need strain relief on Z-gantry...

nweaver, 2023-03-07:
The connecting wires between the Z-gantry and the head itself are under a fair deal of stress:
In particular, if the connection breaks (yellow wire in my case) the tool head itself gets very hot and deforms the z-gantry.

lucian151, 2023-03-25:
Hi @nweaver, thank you for reporting this, we're taking a serious look at how to incorporate your feedback into machines moving forward. Thinking this will take the form of zip-tie mounting features on z-gantry print.

RRudi92, 2023-07-28:
[image]
Happened to me too. Tried to fix it and strengthen it with Hotmelt, but it broke again. I had to Disable N2.

RRudi92, 2023-07-29:
Suggestion: 
Does the Z-Axis NEED over 30mm of travel (Overall travel 60mm)? Most Parts that will be P'n'P'd with such a mashine will be rather flat. Maybe TO Mosfet or the like.
Mounting the Heads lower (e.g. 15mm over PCB-Hight) will reduce travel distance and strain on the cabling.

closed 2023-09-06 by lucian151