opulo-inc/lumenpnp #660: Gantry slams into sides. How to add limit switches

caspersauce, 2023-03-10:
Anyone know how to enable limit switches with the v3 build? I have finally had enough of the gantry smashing into the side when it resets and homes.

I mounted some and wired them up, but it looks like the firmware completely ignores them.

caspersauce, 2023-03-12:
Update... I found this in the v3.0.0 release text:

> The Diag pins on the X and Y axis TMC stepper drivers have been connected to the X and Y limit switch pins. All axis have the support for sensorless homing using a configuration jumper on the back side of the board

I cut the jumper and soldered it to the limit side, and plugged in the limit switches. It now acts like it has already arrived at home when it receives the "G28" command. 

My assumption was then that the "senseless homing" was baked into the firmware and it would take more then just a jumper change. I found this section in the v2.1.2 release of marlin:

```
#define SENSORLESS_HOMING // StallGuard capable drivers only

  #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY 60
    #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY 50
    #define Y2_STALL_SENSITIVITY Y_STALL_SENSITIVITY
```

I commented all this out to disable sensorless homing, then followed the instructions [here](https://docs.opulo.io/byop/motherboard/update-firmware/)

The device now jumps 6 inches as once and slams into the side when the "G28" command is sent.

Does anyone know the proper way to get this to use traditional limit switches? It really feels like this sensorless homing idea is not ready for primetime...

caspersauce, 2023-03-13:
Figured it out...

This is my first adventure in the the world of Marlin. This code located in configuration.h:

```
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true  // Set to true to invert the logic of the endstop.
```

X and Y are not inverted for some reason, so if you use similar endstops to the z axis, it will think it is constantly triggered. If you change this to true for x and y as well as commenting out all the sensorless homing stuff and clipping the jumper on the bottom of the board, the limit switches will work again.

I looked at the schematic and it looks like both Y1 and Y2 are being driven by the same drivers. I can't imagine how the stall-guard performs properly when it has two motors stalling at the same time on the same driver (or possibility at different times due to the gantry not being square.) I honestly didn't know the drivers could drive two motors in parallel. The PWM current waveform they use must go all over the place trying to balance both loads.

If anyone has a similar issue and wants to avoid the sad sound of motors stalling follow the linked instructions and edit the code!

sphawes, 2024-02-29:
thanks for documenting this @caspersauce! there's a new cut of firmware that greatly improves sensorless for v3 machines as well (pull from the latest release).

closed 2024-02-29 by sphawes