opulo-inc/lumenpnp #661: Making jigs open source

JudgeBeeb, 2023-03-12:
Is there any possibility that the bottom-camera-focus-jig and top-camera-focus-jig could be made open source and published on GitHub (and maybe some of the other jigs such as z-belt-alignment-jig)?

Jnesselr, 2023-06-17:
Agreed, this would be awesome and I just ran into it myself. If they are published, it may be useful to put the first mention of the jig as a link on ohai.

asm308, 2024-05-31:
Has there been any movement on this request yet?  I was just looking through the ohai site
https://ohai.opulo.io/lumen/x-gantry/#top-camera-assembly-preparation
and it shows a top camera jig but I’m not able to find this or the bottom camera one.  Are these going to be made available especially since they are in the assembly instructions.
Thanks

stellardynamix, 2026-09-23:
Any update on these?