opulo-inc/lumenpnp #665: Motherboard has no inrush current protection

sphawes, 2023-05-12:
### Please provide information about the board with the issue:
The REV04 motherboard has no method of protecting against inrush current
 
### Talk about the issue you're having here: 
Community member @warasilapm has confirmed that when many feeders are attached to a LumenPnP, on boot the inrush current is quite high, peaking at 9A. The power supply enters overcurrent protection and reattempts every 70ms to power the board. This means on boot, a smaller number of feeders can already be mounted to the machine.

This capture shows the high inrush in blue as the power supply ramps the voltage in yellow, then entering shutdown after about 2ms.
[image]

This capture is a zoomed out version of the first capture, showing the power supply repeatedly entering shutdown and reattempting.
[image]

### Talk about how you think this should be fixed.
This issue can be solved in a number of different ways. An inline inductor on the power rail on the feeders would help reduce this, along with a better power supply. The root of the issue is arguably that the motherboard does not provide any inrush current protection.

Many of the methods recommended in a [similar issue in the feeder repo](https://github.com/opulo-inc/feeder/issues/11
) would be applicable. Having a dedicated eFuse IC on the 24v rail is likely the most reliable and catch-all solve.

TPwire, 2023-05-15:
Hello Stephen,
My name is Alberto G., HW engineer in automotive industry. In ours designs is very normal to have 100A inrush current due to big capacitors with low ESR. However, to limit the inrush current to comply with customer requeriment, we designed something like this:

[image]

Q1 is the typical reverse protection and Q2 is the inrush limiter. Q2 will work in the linear mode during the boot. It is very important that Q2 whitstand the pulse, you can check it with a cauer network simulation using the thermal impedance of the transistor.
C1&C4 are optionals.

PD: I would like to make your lumenPnP for my hobbyist projects, maybe this summer.

stuartpittaway, 2023-05-16:
@sphawes have you seen the suggestions from @TPwire , looks like a neat simple solution, not requiring many components

sphawes, 2023-05-19:
Thank you for the suggestion @TPwire! This does look like an incredibly clean solution. I'll implement this on our next motherboard spin!

G-Pereira, 2023-08-30:
I am leaning more towards choosing a buck regulator with soft start feature on the feeders like we have here on the motherboard.
At the moment we are using a buck/boost/inverting regulator which we are using as a regular buck so we rather pick one with soft start.

G-Pereira, 2023-09-05:
In conjunction with EMI filtering on the feeders the inrush current is no longer an issue.
This is ongoing investigation

G-Pereira, 2023-10-29:
This has been included in future motherboard release. Thank you!

sphawes, 2024-01-04:
I have spun up a REV05b0 board and this approach does power the 24v rail correctly. I have yet to test inrush, but will update here when I do!