opulo-inc/lumenpnp #666: Powering via USB before 24v can result in incorrect speeds

sphawes, 2023-05-16:
# Version Number

v3.0.5

# Bugfix or Enhancement

Enhancement

# Description

When the REV04 LumenPnP is powered on via USB first, the STM32 comes up and boots Marlin. On boot, Marlin sends a bunch of configuration data to the TMC2209 stepper drivers via UART to set microstepping settings. However, the TMC2209 drivers are not powered via the USB bus, so if the 24v rail is not powered at the time of Marlin booting, the drivers miss this initial config data.

As a result, they're running their default microstep settings, but Marlin assumes they've been set correctly, resulting in incorrect movement speeds and distances. Not powering the drivers via USB was meant to reduce the current requirement of the USB port, but the issue has caused confusion amidst users when the machine does not move correctly.

It's an easy fix to just press reset, or power via 24v before USB, but it should be impossible for the stepper drivers to miss their initial configuration data.

# Suggested Solution

There are two solves here:
1. VBUS from the USB port is not used anywhere on the board, and 24v power is required to talk to Marlin.
2. VBUS powers the STM32 and the drivers' 3.3v rail, allowing the config data to be sent on Marlin boot without 24v.

I'm leaning towards number 1. In some instances, folks will attempt to turn off their machine by removing the barrel jack or unplugging the power supply. But if the board is still connected via USB then Marlin has not shut down, causing confusion. The only reason to have VBUS power the STM32 is to be able to test Gcode commands without a 24v power supply, but this is a pretty weak reason in contrast to the confusion that it causes.

Would love any thoughts on this issue!

juantinazzo, 2023-05-22:
What about a reset to the uC when the 24v line goes high? I think it's useful to have te option to talk to the board without having all the "high power" stuff connected. 

Here is a simple circuit for that:

[image]

It uses a few passives and a transistor. It's immune to the PSU noise, while other power-on reset circuits are not.  The difference between C1 and C4 determines the minimum voltage that the reset pulse reaches. With C1 at 100nF it reaches 0.3V, and at 200nF it reaches around 0.15V. I think that should be enough to reset the uC, haven't checked the datasheet. R1 is there to avoid any unwanted reset with 24V line floating, and C2 forms a low pass filter in case there is some weird transition on the PSU power-on, thou it may be ommited.

In the simulation the 24V line has a 1V peak 100kHz sine wave mounted on it to test noise immunity.

G-Pereira, 2023-08-30:
I am leaning more towards removing the option to power via USB as it results in a simpler design.

G-Pereira, 2023-10-29:
Solved by removing option to power via USB

sphawes, 2024-01-04:
Leaving a DNP footprint on the board for a diode that allows VBUS to power the 5v rail, but it should _only_ be used by devs not connecting any high-power 5v peripherals to their board. This has been tested on REV05b0 and works as expected! Will close upon REV05 release.