opulo-inc/lumenpnp #669: Z-Axis Homing must be first on Homing

RRudi92, 2023-06-27:
# Version Number
Marlin bugfix-2.1.x(May 19 2023)

# Bugfix or Enhancement
Bugfix

# Description
The Homing Command issues the machine to home first X, than Y and Z last. 
If the Z-Axis is not in a safe position -> CRASH!!! (This happened to me at least 5 times now, and the machine is only a few days here)

# Suggested Solution
Z **MUST** be homed first. G28Z Command does only issuing a whole Home Command. G28X or G28Y work as expected and only home the respective axis.

RRudi92, 2023-07-06:
In Addition, it might be a good Idea, to do a Z-Axis homing after Nozzle-Change. Best would be a warning, if Z has lost Steps after Nozzle-Change, but I don't know if this is possible at all.

andrewortman, 2023-09-22:
If I understand correctly homing Z on v3 requires one of the heads to extend all the way down while the other hits the limit switch. This will also cause a crash in the Z direction (although, probably not as serious because of the spring mechanism of the nozzle..) - I found a workaround was to go to the last known midpoint in the HOME_COMMAND gcode under GCodeDriver:

```
G1 Z32;
G28;

```

Use at your own risk, of course

RRudi92, 2023-09-23:
Well, this will be of no use if the z-Axis has lost it's position. 

The good solution would be to use an inductive Sensor for Homing at Midpoint. Depending on the state of the sensor, you can determin in wich direction the axis has to move. 

Another option might be to use the motor torque / current to determin a crash in Z -> revert to about half way, do homing X & Y, and repeat Homing Z.

Another way might be to determin if the machine was disabled correctly (e.g. disconnect button, OpenPnP closed), or if the homing appears out of nowhere -> user saw a crash -> pressed Homing button -> assume Z-Axis lost steps and therefore do Z-Axis Homing first.

sphawes, 2024-02-19:
we very intentionally home Z **after** X and Y, because a nozzle plunging into the pick area can be quite a bit worse for PCBs and components in trays/strips than an X gantry crash.

if we have homed before, @andrewortman's solution works wonderfully as it ensures the Z axis is level before homing. if Z **is not** trusted, marlin thinks that it's at Z0, but it's _actually_ at Z31.5, that `G0 Z31.5` command can cause N2 to crash.

the full solution here is to _know if marlin trusts its Z position_. in marlin source, it becomes a "trusted axis" which effectively means it's been homed before. i've been looking into a way to pull this out via serial so openpnp could prompt a user to manually level the nozzles if Z is untrusted.