opulo-inc/lumenpnp #670: Motors are disabled after a way too short time

RRudi92, 2023-06-27:
# Version Number
Marlin bugfix-2.1.x (May 19 2023)
2023-04-05_08-24-36.0aa4ae8

# Bugfix or Enhancement
Bugfix / Enhancement

# Description
The Motors are disabled after a few Minutes. A Position loose is possible.

# Suggested Solution
The Motors should NEVER be disabled, as long as the machine is active. Sometimes you have to do a short break while P'n'Ping, and suddenly you have lost a step or to (without recognizing)

wolfstrassen, 2023-06-28:
I second this. Loosing a step could be critical when doing micro bga ICs

wolfstrassen, 2023-06-30:
Did you try setting up the time with M18/M84?

RRudi92, 2023-07-06:
Yes, M84 S5000 did the trick. It seems to be persistent saved on the machine.

sphawes, 2024-02-18:
this has been fixed in the v3.2 release!

closed 2024-02-18 by sphawes