opulo-inc/lumenpnp #681: Unable to compile the latest Marin using the configuration file

84864797, 2023-09-08:
Marlin-2.1.2.1
Marlin-bugfix-2.1.x
I've tried it all
Unable to use downloaded Lumen_ REV4 Configuration File Compilation Firmware
Can you provide the uncompiled Marlin firmware?

stuartpittaway, 2023-09-08:
Hi @84864797, can you provide some more detail - what errors did you receive, screenshots etc?
Which REV4 configuration source files are you using?
Thanks.

84864797, 2023-09-09:
hello
The configuration file used is Marlin bugfix 2.1. x
Now it can compile normally，But after brushing in the firmware, connecting to the computer shows an unknown USB。
I tried v3 lumenpnp firmware feeder support. bin firmware again, and after connecting to the computer, it displayed an unknown USB device