opulo-inc/lumenpnp #686: Drag Chain specified in 3.0.5 BOM has wrong mounting foot print

rmhorwitz, 2023-10-02:
# Version Number
3.0.5

# Bugfix or Enhancement
This is a BOM related issue.  Component in BOM 3.0.5 is not compatible with the specified design.

# Description
The Drag Chain specified in BOM 3.0.5 vendor's list does not fit either X-Gantry Back or Front-Drag-Chain-Mount.

# Suggested Solution
1) create a mounting adapter plate for the drag chain
2) modify drag chain to fit into existing mounting holes
3) Find correct drag chain with the same hole pattern as the printed parts.
4) Redesign X-Gantry Back and front-drag-chain-mount to use a generic mounting

lucian151, 2023-11-24:
I've added a note to the BOM in commit https://github.com/opulo-inc/lumenpnp/commit/b62a0dfe33a0e547b7dc483e15267243ab76cab2 to alert builders to this issue. This will be ultimately solved by replacing the Amazon source with something else like what LDO is able to supply Opulo.

lucian151, 2024-05-22:
If anyone is still affected by this issue you can make a drill jig for reworking a drag chain's hole pattern. They look like this:
[image]
[image]

You can download the FreeCAD model of these jigs [here](https://cdn.shopify.com/s/files/1/0570/4256/7355/files/drag-chain-drill-jig-baseplate.FCStd?v=1716417490).

closed 2024-05-22 by lucian151