opulo-inc/lumenpnp #692: Printed Parts Revision Information

AzureOz, 2023-11-28:
# Version Number
3.1

# Bugfix or Enhancement
Enhancement

# Description
Currently the BOM file and associated documents list all the printed parts for the current version.  There is no way to know which parts have not changed since a previous version when considering making an upgrade to and existing lumenpnp.

# Suggested Solution
Add a field (or add to an existing field) to indicate the version this part started from.  Then as new versions and revisions are made available it is clear which parts have changed when looking at the parts to decide which parts need reprinting to implement a single or multiple version upgrade to an existing machine.

sphawes, 2024-02-18:
im hesitant to implement this as it's a lot of info to keep track of, and isn't "single source of truth". we currently write release notes talking about what parts changed, but we could be more specific about which parts are worth upgrading. git also allows us to run a diff between two versions, which shows which files have been updated automatically. an example is here: https://github.com/opulo-inc/lumenpnp/compare/v3.1.2...v3.2.0

@lucian151 what do you think about this? we use git so that this kind of thing is easy, i think tracking this is redundant

AzureOz, 2024-02-19:
If that is hard to do then a note, comment or field to say a part is "new" to that revision would be useful.  Then anyone can see what has changed in the new bom compared to their build.