opulo-inc/lumenpnp #698: Electromagnetic Compatibility

sphawes, 2024-01-04:
# Version Number
REV04

# Bugfix or Enhancement
Enhancement

# Description
The LumenPnP motherboard could see improvements to it's electromagnetic compatibility.

# Suggested Solution
Thorough consideration for EMC issues (please leave a comment with specifics if you have thoughts in mind!):

- Conducted and Radiated Immunity - Is the motherboard susceptible to conducted or radiated fluctuations? (ESD, Mains fluctuations, etc)
  - TVS diodes at every exposed point for ESD protection
  - Proper decoupling and smooth start on the 24v rail for the least possible instantaneous draw on the 24V external power supply.
- Conducted and Radiated Emissions - Is the motherboard kicking out any type of conducted or radiated emissions that should be mitigated? (Buck converter / other high speed signal RF noise)
  - Making all switched buck converter traces as short as possible.

sphawes, 2024-01-04:
The REV05b0 board has many of these considerations taken into account, but no doubt that they could be improved!

- TVS diodes have been placed on all exposed connections:
  - Pump/valve connections
  - USB
  - RS-485
  - 24v input from power supply
  - BOOT/RESET pins exposed in AUX header
  - Ring light connectors: data lines and 5V rail
- Stepper drivers have ferrite beads and decoupling caps on every output
- Buck converter traces are kept short (still could be optimized)
- Reverse polarity protection and inrush current protection on VIN

Would love any and all feedback on how these could be made better, plus things that I'm missing!

dan-gies, 2024-01-04:
# Copper-pour "fingers"/"antennae"
Whenever possible, avoid copper structures that are only connected at one end. These can cause problems for radiated immunity and emissions.
[image]
[image]
[image]
[image]
etc...

dan-gies, 2024-01-04:
# Signal return paths
Since this is a 4-layer board with power planes on the inner layers, signals are constantly hopping between reference planes (GND and +3.3V). This can impact both signal integrity and EMC, since it is tough to predict the signal return-current paths. When signal return currents take a long "roundabout" path, it creates an inductive loop, deteriorating signal quality, potentially radiating emissions, and potentially becoming susceptible to incoming noise.

### One example of signals transitioning from top to bottom layers, where the signals have two different return paths that aren't coupled together
[image]

It's easy to brush this off as "not of concern since it's not high-speed", but the STM32F407 _is_ actually capable of decently high rise/fall-times, resulting in high frequency signal content. From the datasheet (below), rise/fall times can be set to as high as 2.5 ns, which equates to roughly a signal bandwidth of 140 MHz. This is territory where return paths start to really matter.
[image]

# Recommendations
## Improve return paths
### Option 1: Use GND planes exclusively on both inner layers, route +3.3V on top/bottom
This may not be ideal, but it's the easiest way to avoid problems. If this is not feasible, you could add more layers, or move to the other option.

### Option 2: Add a high-pass capacitor between +3.3V / GND all over
These capacitors should be as close as possible to **every** grouping of signal layer transitions. The exact value can be determined to pass up to 200 MHz or so. This reduces the inductive loop area; short return paths are good.

## Reduce rise/fall times whenever possible
The STM32F407 has registers to set rise/fall times (AKA "drive strength"). Use the lowest drive strength possible while maintaining good signal integrity (too low means signals aren't reaching Voh/Vol in time, depending on your bus speeds).

JacobChrist, 2024-01-05:
Your probably going to have trouble passing emissions <1GHz with the drivers so far from the motors.  You need to couple the driver to the motor and use a shielded step/direction wires to get the signals to the drivers.  It might be possible but I'm no an RF expert.  In my previous journeys we even purchased NEMA17 motors that have shielded leads (really expensive) and still could not pass being only 30cm from the driver in a conductive housing (with big holes).

JacobChrist, 2024-01-05:
Also, not sure how many layers your board is, but I would go with no less than 4 layers and make sure you have a solid ground plane under all signal traces and that any switching traces are as far from the board edge as possible (lots of ground plane between the board edge and the switching trace).