opulo-inc/lumenpnp #699: Proper Motherboard Enclosure

sphawes, 2024-01-04:
# Version Number
REV04 mobo (v3.1.2 of the Lumen) 

# Bugfix or Enhancement
Enhancement

# Description
The current method of physically protecting the motherboard from damage or ESD is a bottom cover to prevent parts from falling onto it, and a top cover to prevent mechanical damage. This should be improved for better ESD and mechanical protection.

# Suggested Solution
The motherboard and pneumatic components should be fully enclosed in a printed case. This includes moving the motherboard and pneumatics to the left side of the machine, mounted to the bottom rail of the Left Y Gantry, clearing space on the staging plate for an easier time mounting strip feeders and board mounting hardware.

This extra layer of protection should help greatly with ESD concerns. I doubt PLA is the most ideal choice, but it is accessible to builders. This is a strategy that the Prusa Mini uses for its controller enclosure. Any thoughts on the efficacy of PLA in this role is greatly appreciated!

Here's a rough mockup of the idea:

[image]

lucian151, 2024-03-13:
[image]

lucian151, 2024-03-13:
[image]

G-Pereira, 2024-05-04:
Addressed on v4

closed 2024-05-04 by G-Pereira