opulo-inc/lumenpnp #701: Add PWM Capability to MOSFET outputs (Fan Outputs)

warasilapm, 2024-01-04:
# Version Number
Rev5b0

# Bugfix or Enhancement
This is an enhancement.

# Description
The MOSFET pins (MOS1-4) are currently driven by PE2-5. These are not all PWM capable because they lack the AF mapping to a TIM peripheral. The Gcode commnads for these outputs are the Marlin fan commands, which take an 8 bit duty cycle. This is not strictly necessary, but for the solenoids it may save energy (and thereby extend the life of the parts due to lower heat) to PWM them once they are actuated rather than leaving them at full current. This is relatively common.

# Suggested Solution
Move MOS1 and MOS2 to PB8 and PB9 (TIM4/10/11) and move MOS3 and MOS4 to PE5 and PE6 (TIM9). This change to pin assignment. should be fairly straightforward.

Alternatively, just moving the valve drivers to PE5/6 should be sufficient.

Interestingly, the best timer mapping is probably the two valves on PE5/6 since they will share frequency requirements and the pumps  on PB8 and PB9 since they map to a different TIM block for whatever frequency requirements might be needed there.

G-Pereira, 2024-05-04:
This can make sense. But we did not add this capability on purpoose since we didn't felt necessary initially.
@sphawes if you choose to implement this don't forget to check if the mosfet gate current does not violate the MCU pin max current.

warasilapm, 2024-05-06:
> ...if you choose to implement this don't forget to check if the mosfet gate current does not violate the MCU pin max current.

Across that 120 ohm resistor there would be 27.5 mA into the empty gate capacitance. This is just over the per-pin absolute maximum in the datasheet.
[image]
This is quite east to solve with an increased series resistance. On the other hand, that may not be all that relevant with OSPEED set to the lowest value.
[image]
Since the gate capacitance is relatively small, a simply ramping of the output may be sufficient to prevent this issue.

Also, remember that if this _were_ an issue it already exists in the current design. It will just be exacerbated by the move to PWM.

warasilapm, 2024-05-06:
Spice supports my assumption about the slower output speed being sufficient:
[image]
Even at the fastest speed this doesn't quite reach the current limit:
[image]
[image]

warasilapm, 2024-05-06:
Whoops! I did it with an order of magnitude less on the gate capacitance. Looks like it's fine on the slowest output speed but might not be at the highest. My mistake!
[image]
[image]

That said, I have a distant memory of the outputs being current limited so it may not matter at all; the difference in rise/fall times depending on the capacitance would suggest that. It's hard to say though and I can't find any documentation to support this memory at the moment.

warasilapm, 2024-05-06:
As a final note, it looks like the slow speed is the only meaningful one of note:
[image]

Spice here: [mos_gpio_speed.txt](https://github.com/opulo-inc/lumenpnp/files/15224091/mos_gpio_speed.txt)