opulo-inc/lumenpnp #705: Remove I2C MUX and use 2nd I2C Port for Vacuum Sensors

KC0D, 2024-01-04:
# Version Number
REV05

# Bugfix or Enhancement
Enhancement

# Description
The vacuum sensors use the same I2C address so a I2C MUX is used to address each one independently through a single I2C port.  The I2C MUX is a fairly expensive part. 

# Suggested Solution
Remove the I2C MUX and use a second I2C port on the processor to address the 2nd vacuum sensor.  I2C3 (PA8 & PA9) could be used for the 2nd I2C port.  These pins are not currently used.  Removing the MUX saves the cost of the MUX, plus five 10K resistors, two 4.7K resistors and maybe one .1uF cap.    

Marlin would need to be modified to handle this new configuration.

stuartpittaway, 2024-01-08:
I agree removing the MUX would be a benefit.

Unfortunately, Marlin has poor support for i2c as it currently stands - and expanding this to add a second i2c channel would require a lot of development work and testing in the Marlin code base.

There isn't a guarantee that the Marlin project owners would accept this change either.

sphawes, 2024-01-16:
Yeah, unfortunately this is quite a thing to implement. For now, the mux is the simplest solution. I believe that's why we ended up choosing the mux to begin with. Unfortunately the vacuum sensor does not allow for selectable addresses, which would fix this issue as well.

Thanks for the thoughts @KC0D!