opulo-inc/lumenpnp #706: Use USB specific ESD protection package

warasilapm, 2024-01-04:
# Version Number
Rev5b0

# Bugfix or Enhancement
Both?

# Description
The USB lines to/from the ESD diode chosen (SRV05-4) is routed a little awkwardly; I think this is a result of the four line configuration of the part chosen.

# Suggested Solution
Using something like the [USBLC6-2](https://www.st.com/en/protections-and-emi-filters/usblc6-2.html) much simplifies this routing.
[image]

dan-gies, 2024-01-04:
I agree that the routing path on the current part is a little awkward.

@warasilapm note that the suggested part should be routed to only one of the two pins (e.g. 3 or 4) and pass under the part to maintain signal integrity. From the datasheet, ST shows one trace of the diff-pair passing under the part, not through it.

[image]

warasilapm, 2024-01-04:
> I agree that the routing path on the current part is a little awkward.
>
>
>
> @warasilapm note that the suggested part should be routed to only one of the two pins (e.g. 3 or 4) and pass under the part [...]

I looked at that part of the datasheet; that is referring to the VBUS line specifically as it continues towards the rest of the board. If anything, it implies the  optional diode should follow the ESD IC in the circuit rather than branching off at the connector pin. IIRC the screen cap you took makes sense given a micro-b pinout but I didn't check.

The recommended layout reflects how I show the data lines.

dan-gies, 2024-01-04:
@warasilapm you're right! I rushed to conclusions.

warasilapm, 2024-01-04:
Image for reference:
[image]

Btw I checked and it definitely did not make sense for micro USB. Who knows what they are thinking with that VBUS trace.

dan-gies, 2024-01-04:
For both these parts (existing and new) it's a little weird that they pull the diff-pair so far apart, but I guess the ±Ω tolerance is big enough that it doesn't really matter for FS speeds. For 2.0 I'd think it'd be sub-par, but that's irrelevant here.