opulo-inc/lumenpnp #708: Add reverse-current protection to supplies

dan-gies, 2024-01-06:
# Version Number
REV05

# Bugfix or Enhancement
Enhancement

# Description
When powering down, you can get reverse current through the intrinsic body diodes of the MOSFETs within the LDO and buck parts (if Vin depletes faster than Vout, for each part respectively).

# Suggested Solution
Best practice is to add reverse-current protection, or test that Vin _always_ discharges faster than Vout for each part. In my view, adding the reverse current protection is low-hanging fruit to improve reliability. My preference is to add a Schottky diode in reverse (anode at Vout, cathode at Vin) as shown in Figure 3 in this [TI appnote](https://www.ti.com/document-viewer/lit/html/SSZT658), as it doesn't reduce efficiency during normal operation.

[image]

sphawes, 2024-01-16:
Thanks for the note @dan-gies! That makes sense to me, I'll add a Schottky to the LDO.

I couldn't find an application note about the buck converter requiring one as well. I can't imagine how adding it could cause issues if during normal operation it's constantly reverse-biased. I'll add it to the next pass of the board, but anything you have regarding it's usefulness for the buck would be helpful to see!

Thanks for the thorough review!

warasilapm, 2024-01-25:
> I couldn't find an application note about the buck converter requiring one as well. I can't imagine how adding it could cause issues if during normal operation it's constantly reverse-biased. I'll [...]

This is the correct solution for the buck regulator as well. A Schottky from 5V to VDC is the most straightforward way to go.

However, I don't think this is strictly necessary. The AP1117 is a BJT device which doesn't have the body diode issue addressed in the TI appnote. I know some linear regulators are very sensitive to reverse current but I don't think this LDO will have any issues. For the 5V switching regulator, there is so much capacitance on VDC and sufficient current demand (by way of 3V3 and the LEDs if they are on) that 5V should always discharge faster than VDC. You can run a test to verify this. I think the current design does not suffer from this type of failure because by the time the regulators would have reverse current the voltages are all around a volt or so (when the ICs can no longer run) so the currents are too small to cause damage to the devices.

G-Pereira, 2024-05-04:
This a good point however in this design our capacitance in the 3.3V rail is small so I don't see the need for the diode. This LDO is not a sensitive one also.
We could discuss this @sphawes if you'd like